ControlSystemStudio/phoebus #2520: Alarm severity and status cleared when config is modified

georgweiss, 2023-01-25:
This seems to be a bug, I think:

- The alarm table (and other views?) shows a latching alarm that has changed severity and status, e.g. from major to ok.
- User open config dialog and updates for instance description
- User closes config dialog -> alarm severity and status switch to the current severity and status. Next alarm that is not ok will update the view as expected.

kasemir, 2023-01-25:
Well, when the alarm server restarts, it starts over with all alarms at their current severity. A previously latched severity is lost.

Same when updating the config of one channel: That channel starts over with its current severity.

georgweiss, 2023-01-25:
OK, expected behavior in other words...

kasemir, 2023-01-25:
Frankly, starting over with a changed channel or all channels for a server restart was much easier to implement.
In a way it's also comforting because you see exactly how each channel behaves. It will filter, annunciate, latch, ..
At the end of a maintenance period, that's usually good:
Restart the alarm server, see all the current alarms appear with annunciation etc.
In the alarm table, enter "*" for the pattern to select them all, press "Acknowledge", and everybody's convinced that we're ready to operate.

Same with a changed channel: After you change something, you see if it works as intended because the current alarm will come in again etc.

There is one unfortunate part: If you have some channels configured to only appear with a 5 or even 15 minute delay, you'll see them appear 5 or 15 minutes after the restart. Again this can be seen as confirming that the filter/delay works, but it's somewhat of a nuisance that you need to press "Acknowledge" again after ~5 and then ~15 minutes.

georgweiss, 2023-01-25:
Thanks for the explanation @kasemir .

closed 2023-01-25 by georgweiss