ControlSystemStudio/phoebus #2525: openDisplay() behavior vs openOPI()

mcnanneyd, 2023-01-31:
It seems that the behavior of the new openDisplay() function differs from the former openOPI() behavior from BOY. When used in embedded files, openOPI() would open the screen relative to the parent screen. In Phoebus, it opens the screen relative to the embedded screen itself. 

@tanviash suggested that the issue may be that openOPI() would calculate the absolute path of the file referenced with a relative path. openDisplay() seems to lack the support to calculate the absolute path from the relative path, and requires that an absolute path is given, or the relative path is relative to the embedded opi. 

Does anyone have any thoughts on this behavior? It is becoming a bit of an issue as we are working on converting to cs-studio Phoebus. Thank you!

kasemir, 2023-01-31:
> in embedded files, {openDisplay} opens the screen relative to the embedded screen itself.

That's fundamentally correct. A script or button inside a display opens other displays relative to where it is.
For an embedded display, that's relative to that embedded display.

For example, assume you have this directory layout:
```
somewhere/motor/MotorSummary.bob
somewhere/motor/MotorDetail.bob

somewhere_else/TopLevel.bob
```

TopLevel.bob can embed one ore more MotorSummary.bob files, which in turn have buttons to open MotorDetail.bob
Since the files are resolved relative to where the button/script resides, MotorSummary.bob, it finds MotorDetail.bob in the same folder.

If it were instead trying to locate it relative to the parent display, it would look somewhere_else and be real sad.

In BOY, the layering of embedded displays was lost because the embedded content was merged into one large display model. That also meant that opening a display had to wait until all embedded content was loaded, whereas the display builder can now load embedded content in background threads while the top level content is already shown.
So in BOY the location of the embedded display is somewhat lost, everything runs in the top level display context.

If you want to port scripts from openOPI to openDisplay, I think you can do something like this to resolve relative to the parent display, going up to the desired level:
```
# Go one level up
parent = widget.getParent().get()

# Go two levels up
parent = widget.getParent().get().getParent().get()

# Go up to the top-level display
parent = widget.getTopDisplayModel()

ScriptUtil.openDisplay(parent, "whatever.bob", "REPLACE", None)
```

mcnanneyd, 2023-02-07:
@kasemir - widget.getTopDisplayModel() was exactly what I needed! I put this inside a portable script. It will certainly ease our Phoebus transition. Thank you!

kasemir, 2023-02-07:
Sounds good, closing this.

closed 2023-02-07 by kasemir