ControlSystemStudio/phoebus #2530: Deleting items in the Data Browser leaves behind "empty" scales in the UI for graphs

abrahamwolk, 2023-02-01:
When a PV is added under the "Traces" tab, the graph of its values during the last hour or so is shown in the UI. On the left of the graph there is a scale for the graph.

If the PV is selected in the bottom half of the window, and the option "Delete Item" is clicked, then the PV is deleted from the bottom half of the window, the graph of its historic values is removed, and the text along the scale for the graph is also removed. However, the scale itself is not removed.

The following screenshot shows the result of adding five PVs and removing them using "Delete Item". On the left, there are five "empty" scales:

[image]

Shouldn't these empty scales be removed when deleting PVs from the Data Browser?

kasemir, 2023-02-01:
> Shouldn't these empty scales be removed when deleting PVs from the Data Browser?

Users can configure an axis by setting its color, min/max, log axis or not, grid or not, on right or left, and then keep  adding or removing traces.
When they remove the last trace, we keep the axis in case they are about to add/move traces to that axis, so they don't have to start over with creating the axis.

If you do want to remove all unused axes, use the context menu entry labeled "Remove Empty Axes" which has been provided for exactly this use case.

[image]

kasemir, 2023-02-01:
Similarly, we don't close the data browser when you delete the last trace, nor do we power the computer down when you close CS-Studio.

abrahamwolk, 2023-02-01:
Alright!

closed 2023-02-01 by abrahamwolk