ControlSystemStudio/phoebus #2535: Setting es_sniff to false doesn't disable auto-configured Elasticsearch sniffer

jbellister-slac, 2023-02-07:
### Context

When doing a deployment of the latest 4.7.1 version of the alarm-logger, this error was logged every 5 minutes despite leaving es_sniff to the default of false:

```
Feb 06, 2023 4:57:10 PM org.elasticsearch.client.sniff.Sniffer$Task run
SEVERE: error while sniffing nodes
java.net.ConnectException: Connection refused
	at org.elasticsearch.client.RestClient.extractAndWrapCause(RestClient.java:918)
	at org.elasticsearch.client.RestClient.performRequest(RestClient.java:299)
	at org.elasticsearch.client.RestClient.performRequest(RestClient.java:287)
	at org.elasticsearch.client.sniff.ElasticsearchNodesSniffer.sniff(ElasticsearchNodesSniffer.java:106)
	at org.elasticsearch.client.sniff.Sniffer.sniff(Sniffer.java:209)
	at org.elasticsearch.client.sniff.Sniffer$Task.run(Sniffer.java:140)
	at org.elasticsearch.client.sniff.Sniffer$1.run(Sniffer.java:81)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: java.net.ConnectException: Connection refused
	at java.base/sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at java.base/sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:777)
	at org.apache.http.impl.nio.reactor.DefaultConnectingIOReactor.processEvent(DefaultConnectingIOReactor.java:174)
	at org.apache.http.impl.nio.reactor.DefaultConnectingIOReactor.processEvents(DefaultConnectingIOReactor.java:148)
	at org.apache.http.impl.nio.reactor.AbstractMultiworkerIOReactor.execute(AbstractMultiworkerIOReactor.java:351)
	at org.apache.http.impl.nio.conn.PoolingNHttpClientConnectionManager.execute(PoolingNHttpClientConnectionManager.java:221)
	at org.apache.http.impl.nio.client.CloseableHttpAsyncClientBase$1.run(CloseableHttpAsyncClientBase.java:64)
	... 1 more
```

In my case, the alarm-logger and elastic cluster are on different machines, and I don't want sniffing as the load balancing will be handled elsewhere.

But to more simply reproduce this, can just build the alarm-logger from source and run it without pointing it to any elasticsearch. This fails of course, but the sniffer error will still be part of the failure despite trying to leave it off.

### Workaround

It looks like the reason for this is that the AlarmLoggingService is annotated with `@SpringBootApplication` causing spring to auto-configure an elasticsearch client with sniffer active based on the sniffer dependency:

https://docs.spring.io/spring-boot/docs/3.0.1/reference/html/data.html#data.nosql.elasticsearch.connecting-using-rest.restclient

I fixed this by excluding that specific auto-configuration by adding this to the `application.properties ` of the alarm-logger, after which the logs are clean and the alarm-logger still functions fine:

`spring.autoconfigure.exclude=org.springframework.boot.autoconfigure.elasticsearch.ElasticsearchRestClientAutoConfiguration`

Maybe this should just be the default since the elastic client which is used for alarm logging is manually configured?

georgweiss, 2023-02-07:
I guess that if we disable the sniffer by default, in a clustered environment it would have to be explicitly enabled, right? If so, then we chose between suppressing log messages and making the default configuration suited for a clustered Elastic setup.

jbellister-slac, 2023-02-07:
From what I can tell, there are actually 2 elastic clients created. One that I don't think is actually used anywhere? - the autoconfigured one created by spring boot which I am excluding. And then the manually configured one which does all the work, created in `ElasticClientHelper` with optionally activated sniffer here:

https://github.com/ControlSystemStudio/phoebus/blob/44e3c31f642f89fb69759b169682a36df3124e92/services/alarm-logger/src/main/java/org/phoebus/alarm/logging/ElasticClientHelper.java#L91-L103

So I think by turning off the auto-configuration I'm not actually changing anything functionality wise because all that client/sniffer is doing is running a task in the background that isn't used anywhere. And setting es_sniff to true will get that manually created sniffer running even after my change.

I think the other option would be to switch over to using the spring boot initialized one, though that would be additional work requiring a re-write of `ElasticClientHelper`.

jbellister-slac, 2023-02-07:
The spring boot one is created from this dependency:

https://github.com/ControlSystemStudio/phoebus/blob/44e3c31f642f89fb69759b169682a36df3124e92/services/alarm-logger/pom.xml#L60-L69

And sniffer task unaffected by `es_sniff` runs immediately after

https://github.com/ControlSystemStudio/phoebus/blob/44e3c31f642f89fb69759b169682a36df3124e92/services/alarm-logger/src/main/java/org/phoebus/alarm/logging/AlarmLoggingService.java#L202

shroffk, 2023-05-23:
@jbellister-slac thanks for figuring this out and fixing this.
I think your initial solution of disabling the auto configured es client is the best way forward currently.

In an ideal situation we would use the elastic clients created by springboot... however, I think we tend to have less control over how the upgrades are incorporated into springboot and elastic does change quite a bit between releases. So the more ugly but more rigidly controlled creation of elastic clients works better from a controls system use case.