ControlSystemStudio/phoebus #2545: alarm-server does weird reverse IP lookups, breaking the connection

MichaelRitzert, 2023-02-13:
I'm trying to start the alarm server in a docker environment.

Usually, this should be as easy as using -server IP:9092, where IP is the IP address of the kafka server.
However, when I try this, the IP address is reverse-resolved to the docker-internal host name, e.g. c7c24f5b62e7:9092. In the end, c7c24f5b62e7 cannot be resolved to an IP address and the connection fails. This is quite annoying since I started with an IP address in the first place to prevent exactly this from happening.

For my test environment, I added c7c24f5b62e7 to /etc/hosts, but of course this is not an option for the production environment where the name will change.

I'm not sure who is to blame — the kafka client libraries or the alarm server use of them, but anyhow: Are there any suggestions to make sure the IP is used as given on the command line?

georgweiss, 2023-02-14:
@andersharrisson-ess, would you be able to comment on this?

andersharrisson-ess, 2023-02-14:
Hi,
we are not running the alarm server nor kafka in Docker.
What I think you are experiencing is the Kafka advertised brokers. When a client requests a connection with Kafka, it gets a response from Kafka with the available brokers in the Kafka cluster. Even for single node Kafka instances this is true. And, if I am not mistaken, Kafka by default "advertise" its hostname as broker-uri. This can be configured though, so please look into the Kafka documentation and find how to set the advertised broker addresses to the IP and hopefully it will solve the issue.
Good luck.

MichaelRitzert, 2023-02-14:
OK, I figured it out. After I increasing the log level, I could see that it's not on the first connection a I thought, but actually returned by the automatic discovery. I had hoped that this is off if I just don't configure it.
I have now added a kafkaproxy instance to adjust the IP addresses in the protocol between the internal and external networks.

PS: Ine thing I learned: kcat which I used for debugging does not care about all of that.

closed 2023-02-14 by MichaelRitzert