ControlSystemStudio/phoebus #2581: Save and Restore - "Create/add to a Configuration" option not working as expected

UrbanBobek, 2023-03-13:
Hi all,

I am getting errors when using the `Create/add to a Configuration` option in Phoebus. I am using the latest version of Phoebus and S&R. 

I right click on a widget that is using a certain PV and select the `Create/add to a Configuration` option and a new window appears. If I click on the `Browse` button nothing happens and the following error appears in the CLI:
```
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException: Cannot invoke "org.phoebus.applications.saveandrestore.ui.SaveAndRestoreController.matchesFilter(org.phoebus.applications.saveandrestore.model.Node)" because "this.saveAndRestoreController" is null
        at org.phoebus.applications.saveandrestore.ui.BrowserTreeCell.updateItem(BrowserTreeCell.java:155)
        at org.phoebus.applications.saveandrestore.ui.BrowserTreeCell.updateItem(BrowserTreeCell.java:52)
        at javafx.scene.control.TreeCell.updateItem(TreeCell.java:533)
        at javafx.scene.control.TreeCell.indexChanged(TreeCell.java:490)
        at javafx.scene.control.IndexedCell$1.invalidated(IndexedCell.java:85)
        at javafx.beans.property.IntegerPropertyBase.markInvalid(IntegerPropertyBase.java:113)
        at javafx.beans.property.IntegerPropertyBase.set(IntegerPropertyBase.java:148)
        at javafx.scene.control.IndexedCell.updateIndex(IndexedCell.java:130)
        at javafx.scene.control.skin.VirtualFlow.setCellIndex(VirtualFlow.java:1804)
        at javafx.scene.control.skin.VirtualFlow.getCell(VirtualFlow.java:1788)
        at javafx.scene.control.skin.VirtualFlow.getOrCreateCellSize(VirtualFlow.java:3026)
        at javafx.scene.control.skin.VirtualFlow.getOrCreateCellSize(VirtualFlow.java:2998)
        at javafx.scene.control.skin.VirtualFlow.recalculateAndImproveEstimatedSize(VirtualFlow.java:3086)
        at javafx.scene.control.skin.VirtualFlow.recalculateEstimatedSize(VirtualFlow.java:3069)
        at javafx.scene.control.skin.VirtualFlow$5.invalidated(VirtualFlow.java:860)
        at javafx.beans.property.IntegerPropertyBase.markInvalid(IntegerPropertyBase.java:113)
        at javafx.beans.property.IntegerPropertyBase.set(IntegerPropertyBase.java:148)
        at javafx.scene.control.skin.VirtualFlow.setCellCount(VirtualFlow.java:904)
        at javafx.scene.control.skin.TreeViewSkin.updateItemCount(TreeViewSkin.java:396)
        at javafx.scene.control.skin.TreeViewSkin.setRoot(TreeViewSkin.java:379)
        at javafx.scene.control.skin.TreeViewSkin.<init>(TreeViewSkin.java:166)
        at javafx.scene.control.TreeView.createDefaultSkin(TreeView.java:1060)
        at javafx.scene.control.Control.doProcessCSS(Control.java:886)
        at javafx.scene.control.Control$1.doProcessCSS(Control.java:89)
        at com.sun.javafx.scene.control.ControlHelper.processCSSImpl(ControlHelper.java:67)
        at com.sun.javafx.scene.NodeHelper.processCSS(NodeHelper.java:147)
        at javafx.scene.Node.processCSS(Node.java:9603)
        at javafx.scene.Node.processCSS(Node.java:9596)
        at javafx.scene.Scene.doCSSPass(Scene.java:585)
        at javafx.scene.Scene$ScenePulseListener.pulse(Scene.java:2507)
        at com.sun.javafx.tk.Toolkit.lambda$runPulse$2(Toolkit.java:407)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at com.sun.javafx.tk.Toolkit.runPulse(Toolkit.java:406)
        at com.sun.javafx.tk.Toolkit.firePulse(Toolkit.java:436)
        at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:575)
        at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:555)
        at com.sun.javafx.tk.quantum.QuantumToolkit.pulseFromQueue(QuantumToolkit.java:548)
        at com.sun.javafx.tk.quantum.QuantumToolkit.lambda$runToolkit$11(QuantumToolkit.java:352)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(WinApplication.java:184)
        at java.base/java.lang.Thread.run(Thread.java:1589)
```

I am also unable to change the contents of the `Location` field. 

I have a couple of questions regarding the `Create/add to a Configuration`  option. If I use the path that is set by default I can create a new configuration with the PV of the selected widget. However, I cannot add another PV to an already existing configuration. If I write the name of the previous configuration in the `Name` field, it turns red and it disables the `Save` button. How can one add PVs to an already existing configuration using `Create/add to a Configuration` option?
In the S&R documentation, the `Create/add to a Configuration` option is mentioned only when using Channel Table. Is it only meant for usage with Channel Finder?

Any feedback would be greatly appreciated!

Urban

georgweiss, 2023-03-13:
I will look into this.

georgweiss, 2023-03-14:
Work ongoing to fix the dialog, sorry for the inconvenience.

In the meanwhile: the Location should not be editable. It is set by choosing a node in the dialog launched when clicking Browse button.

UrbanBobek, 2023-03-14:
Great, thank you the quick response and the explanation!

georgweiss, 2023-03-16:
@UrbanBobek , can we close this?

UrbanBobek, 2023-03-17:
The fix resolved the issue. Thank you!

closed 2023-03-17 by UrbanBobek