ControlSystemStudio/phoebus #2582: "Disabled" widgets use transparency

kasemir, 2023-03-13:
Widgets that are connected to read-only PVs or for other reasons disabled started out using proper "disabled" JFX widgets, but that has the undesired side effect that the widgets are truly disabled and no longer support a context menu, tool tip, middle-button PV name copy etc.

So instead of disabling the underlying widgets, a "not_enabled" style was used, defined as `-fx-opacity: 0.4`.
This resulted in the widgets looking almost exactly like "disabled" widgets, changing all their original colors to a lighter, "ghosted" version, while still supporting tool tips and context menus which might be used to inspect the PV, determine why it's read-only or disconnected etc.

Now @tanviash noted local older BOY displays with text below TextInput widgets that becomes visible when the TextInput is disabled and thus transparent.

At first glance, fixing this by removing the "not_enabled" style would require updating about 16 widget types. Instead of simply applying the "not_enabled" style, each widget would have to for example adjust its colors to a lighter version or find other ways to represent "disabled".

kasemir, 2023-03-13:
Looks like proper "disabled" widgets do the same based on the "modena.css" snippet shown below.

```
/* ====   DISABLED THINGS   ================================================= */

.label:disabled,
.button:disabled,
.toggle-button:disabled,
.radio-button:disabled,
.check-box:disabled,
.hyperlink:disabled,
.menu-button:disabled,
.split-menu-button:disabled,                                                   
.slider:disabled,
.scroll-bar:disabled,                                                          
.scroll-pane:disabled,
.progress-bar:disabled,
.progress-indicator:disabled,
.text-input:disabled,
.choice-box:disabled,
.combo-box-base:disabled,
.date-cell:disabled > *, /* This is set on children so border is not affected. */
.list-view:disabled,
.tree-view:disabled,
.table-view:disabled,
.tree-table-view:disabled,
/*.tab-pane:disabled,  / Removed for JDK-8144983 */
.titled-pane:disabled > .title,
.accordion:disabled > .titled-pane > .title,
.tab-pane > .tab-header-area > .headers-region > .tab:disabled,
.menu:disabled, 
.menu-item:disabled,
.list-cell:filled:selected:focused:disabled,
.list-cell:filled:selected:disabled,
.tree-cell:filled:selected:focused:disabled,
.tree-cell:filled:selected:disabled,
.tree-cell > .tree-disclosure-node:disabled,
.tree-table-row-cell > .tree-disclosure-node:disabled,
.table-row-cell:selected:disabled,
.tree-table-row-cell:selected:disabled,
.table-cell:selected:disabled,
.tree-table-cell:selected:disabled,
.spinner:disabled {
    -fx-opacity: 0.4;
}
```

cogansco, 2023-03-13:
For most screens it is not a problem, but on others, background colors or background text bleed through the transparent disabled fields in an unexpected way.  I agree the simple fx-opactity CSS tag would have been an elegant solution, and more complicated CSS would be tricky to accomplish for every different type of widget.

Given that, does it make sense to spend more effort to enable the context menus in "properly" disabled JFX widgets?  Or is that just completely impossible?

kasemir, 2023-03-13:
> spend more effort to enable the context menus in "properly" disabled JFX widgets?

That doesn't help at all. "Properly" disabled JFX widgets use `-fx-opacity: 0.4`.
Transparency is thus the standard for disabled widgets, we didn't just make that up on our own. 
We simply use the same `-fx-opacity: 0.4` idea, and add a "No write access" cursor, while leaving the widget enabled so the menu etc. still function.

For the specific example screen, I'd say that's a good lessons-learned for using all the specifics of one tool and then being stuck with that exact behavior, vs. trying to create more generic screens that will work with the next display tool and the one after that. Just a text input with PV works fine, shows the value, alarm and read-only state. But a text input with rules to change color, rule to change visibility, text behind etc. means you then get to re-do that in slightly different ways in the next tool.

Back to the styling, I've looked through https://docs.oracle.com/javafx/2/api/javafx/scene/doc-files/cssref.html if there's another way to make the widget appear "ghosted" via a generic style, but can't find anything. So this would mean updating the ~16 widgets to each do their own version of a "disabled" look, and doing it in a way that differs from the basic JFX idea.

For what it's work, if the next display tool is web-based: The default for HTML items with ":disabled" pseudo-class is also slightly transparent. So that's the industry standard, and it doesn't make much sense to fight that.