ControlSystemStudio/phoebus #2583: Display Builder: last save timestamp and user in BOB file data

pavel-ch, 2023-03-14:
I have got a request from my users:
as not on all projects/sites GIT is available, and not always file timestamps are preserved, they ask to save last edit timestamp and editing user (OS user) to the BOB file itself.
I modeled the function using 2 new readonly string properties in DisplayModel (default "" changed on Save in Editor):
[image]
In the Bob file:
[image]

I know the function is not bulletproof (in case of external text editors/convertors etc. are used). Basically this is not a target; the target is to document last change made by Editor.
Can you consider and decide if you want to merge this or not @kasemir @shroffk ?
If you find it unsuitable feel free to reject; I can live without that as well.
Thank you.

kasemir, 2023-03-14:
For those who do use a version control tool, we've been trying to minimize bogus changes. For example, earlier display tools would simply save the properties in whatever order some hash map held them. Or version updates of the tool would create different files even if the widget settings were really the same.
The time stamp means that even when changing nothing, just opening a file in the editor and saving it again, the file would now have that one changed line.
I also worry what's next: Add options so that only the user who created the file can modify it?

All of the version control can be handled just fine with cvs/svn/mercurial/git, write access can be handled by the file system, so I'd rather not duplicate those ideas.

How about taking one step back and save the user and date within a comment at the start of the file?
```
<?xml version="1.0" encoding="UTF-8"?>
<!-- Saved on ... by ...  -->
<display version="2.0.0">
```

If you do need to know, you can open the file in a text editor, but it doesn't rise up to the level of an actual display property. 
Tools like git will still consider a plain open & save as a change because the date and maybe user differ, but a reviewer can tell that it's a comment, and restore the original version.

pavel-ch, 2023-03-14:
Ok, good point,
I do not want to influence your workflow.
But, if you can tolerate that every Save operation generates a file change (I personally consider the Save operation as a clear intention to save a change), I think this two-lines addition is what you mentioned above:

org.csstudio.display.builder.model.persist.ModelWriter constructor
```
    public ModelWriter(final OutputStream stream) throws Exception
    {
        final XMLStreamWriter base =
            XMLOutputFactory.newInstance().createXMLStreamWriter(stream, XMLUtil.ENCODING);
        writer = new IndentingXMLStreamWriter(base);

        writer.writeStartDocument(XMLUtil.ENCODING, "1.0");
        writer.writeCharacters("\n");
        writer.writeComment("Saved on " + LocalDateTime.now().toString() + " by " + System.getProperty("user.name"));
        writer.writeStartElement(XMLTags.DISPLAY);
        writer.writeAttribute(XMLTags.VERSION, DisplayModel.VERSION.toString());
    }
```

Giving:
[image]

Thank you for go/nogo.

kasemir, 2023-03-14:
Yes, I think that would be fine.

closed 2023-03-27 by kasemir