ControlSystemStudio/phoebus #2588: Save and restore : Compare snapshots

agaget, 2023-03-14:
Still me.
On the last updated commit, I got the following trace when I click on "compare snapshot" (right click on snapshot). I have tried selecting many snapshots.

Service is well update this time

```
2023-03-14 15:50:35 AVERTISSEMENT [org.phoebus.applications.saveandrestore.ui.SaveAndRestoreService] Failed to compare snapshot
java.lang.ClassCastException: class org.phoebus.applications.saveandrestore.ui.search.SearchAndFilterTab cannot be cast to class org.phoebus.applications.saveandrestore.ui.snapshot.SnapshotTab (org.phoebus.applications.saveandrestore.ui.search.SearchAndFilterTab and org.phoebus.applications.saveandrestore.ui.snapshot.SnapshotTab are in unnamed module of loader 'app')
        at org.phoebus.applications.saveandrestore.ui.SaveAndRestoreController.compareSnapshot(SaveAndRestoreController.java:417)
        at org.phoebus.applications.saveandrestore.ui.SaveAndRestoreController.compareSnapshot(SaveAndRestoreController.java:407)
        at org.phoebus.applications.saveandrestore.ui.ContextMenuSnapshot.lambda$new$0(ContextMenuSnapshot.java:56)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:234)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
        at javafx.event.Event.fireEvent(Event.java:198)
        at javafx.scene.control.MenuItem.fire(MenuItem.java:459)
        at com.sun.javafx.scene.control.ContextMenuContent$MenuItemContainer.doSelect(ContextMenuContent.java:1385)
        at com.sun.javafx.scene.control.ContextMenuContent$MenuItemContainer.lambda$createChildren$12(ContextMenuContent.java:1338)
        at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:247)
        at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:234)
        at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
        at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
        at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
        at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
        at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
        at javafx.event.Event.fireEvent(Event.java:198)
        at javafx.scene.Scene$MouseHandler.process(Scene.java:3894)
        at javafx.scene.Scene.processMouseEvent(Scene.java:1887)
        at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2620)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:411)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:301)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$2(GlassViewEventHandler.java:450)
        at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:424)
        at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:449)
        at com.sun.glass.ui.View.handleMouseEvent(View.java:551)
        at com.sun.glass.ui.View.notifyMouse(View.java:937)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:316)
        at java.base/java.lang.Thread.run(Thread.java:829)

```

georgweiss, 2023-03-14:
OK, I see the problem. Will fix a.s.a.p.

Workaround is to make sure the base snapshot (with which you want to compare) is loaded into a snapshot view (tab) and selected as active.

agaget, 2023-03-14:
Ok thanks, the workaround works, for now it's enough for me.

georgweiss, 2023-03-16:
@agaget , can we close this?

agaget, 2023-03-16:
Just tested, it works, I close