ControlSystemStudio/phoebus #2595: Properly recognize unsigned data in intensity plots

daykin, 2023-03-15:
We have AreaDetector-based camera feeds being rendered in Intensity plots, and instead of the incoming data being in the range [0,65535], it is instead [-32768,32767]. As a result, we need to input unusual color maps to accommodate. Is it possible to detect signedness of incoming data from Channel Access?

kasemir, 2023-03-15:
Channel access only has signed data types.

PV Access has both signed and unsigned.

In CSS/Phoebus, you can use PV Access to look at area detector images.
If you use Channel Access, the "Image" plot has an option to treat the data as unsigned since the data would otherwise not tell us.

daykin, 2023-03-15:
Ah, I see that now. Thanks!

closed 2023-03-15 by daykin