ControlSystemStudio/phoebus #2608: Alarm tree display broken as side effect of configurable background color

kasemir, 2023-03-22:
#2449 added configurable background colors based on alarm severity to the alarm tree, alarm table, PV tree etc.

Turns out that broke the alarm tree display.

The tree view had a small triangle to the left of intermediate nodes that allow opening/closing the sub tree, and the background was used to indicate the selected cell when clicking on a cell or traversing the three using cursor keys:

[image]

Now that the tree cell backgrounds are set based on the alarm colors, clicking on "Demo" in the above example will _remove_(!) that expand/collapse triangle, and there is no more visual feedback of the selected item:

[image]

That was actually a known caveat because I guess using `setBackground` had been tried before,
https://github.com/ControlSystemStudio/phoebus/blob/cc28ab95ffe85159242c0a580a820656ab40c4e1/app/alarm/ui/src/main/java/org/phoebus/applications/alarm/ui/tree/AlarmTreeViewCell.java#L33
```
        // Note: Cannot use background because that's used by style sheet and selection/cursor
```

georgweiss, 2023-03-23:
Speaking of latest JavaFX... If you mean v 20 then we need to bring up Java version.

kasemir, 2023-03-23:
I tried this with JFX 20 and there was no difference in the alarm tree.
The modena.css for the tree simply includes changes to the background for the selection/cursor, and it also uses the background for the 'arrow', so setting the cell background clashes with that.
The fix on the branch is to have our own `Label` or `HBox(Icon, Label)` and then you can update the background of that Label while the tree can continue to do what it wants with the overall background.

closed 2023-03-23 by kasemir

kasemir, 2023-03-24:
The missing expand/collapse triangle was minor, what really caused trouble is the missing cell updates,  #2611
I was looking at that when I noticed this issue.

I think it's quite common that you have some section of the alarm tree open with many PVs, only partially visible because the list exceeds the window height, and then those that are not shown end up with the wrong value once scrolled into the viewport. I can reproduce that with a simple demo that suggests `TreeItem.setValue()` simply doesn't always work, you need to replace the `TreeItem`. Fixed that in #2615, but now overall disenchanted with the `TreeView`.

PR #2612 (2023-03-23): Alarm tree fixes
Fixes #2608