ControlSystemStudio/phoebus #2617: X/Y plot widget and shift parameter

mirek23, 2023-03-24:
Hello,

When I changed (shifted by the same offset)  the X min and X max then the plot is shifted but I would like to have in the same position regardless the X min and X Max.
Is it possible to change the X Axis labeling as such that  it does not effect the plot position?
Another option would be to use the parameter to shift the plot by the same offset to follow X min and X max shift.

Best Regards
Mirek

mirek23, 2023-03-30:
Hi Georg,

OK, I will describe the issue differently.
When I setup the X Y widget I just add the WF PV into trace PV Y property. The trace PV X stays empty.
The plot by default start at X=0 with this configuration. 
When I change the X axis property minimum and maximum then the X axis changes accordingly by the desired waveform stays at the initial position ie.  starting at X=0. Is there a way to "shift" i.e add the offset to the WF in X coordinates as such that the waveform follows the X scale min value?
  
Best Regards
Mirek

georgweiss, 2023-03-31:
You could construct an array for the X axis PV like so:
```=arrayOf(0, 1, 2, 3, 4, 5)```
Then if you also need to scale X values you could use:
```=scale(arrayOf(0, 1, 2, 3, 4, 5), scalingFactor, offset)``` where both ```scalingFactor``` and ```offset``` can be PV names.
If on the other hand you need to shift instead of scale X values then X axis PV should probably be:
```=arrayOf(0 + shift, 1 + shift, 2 + shift, 3 + shift, 4 + shift, 5 + shift)``` where ```shift``` can be PV.

Manipulating the X axis min and max can be done in a rule using the same ```scalingFactor``` or ```shift```.