ControlSystemStudio/phoebus #2627: Thermometer widget: request for functionality

b-spruck, 2023-04-05:
Is it on purpose that the thermometer widget does not have any functionality to plot value+unit, axis, nor alarm limits like it was in CSS?  In my feeling this renders it rather useless.
I can simply add a text widget nearby as workaround, but this would not solve axis and limit display.

kasemir, 2023-04-05:
TL;DR Yup

b-spruck, 2023-04-06:
O.k. I started with a bit easier widget, adding limit area indicator to the scale, as used in tank. Like the style as it is used in meters. (more or less copy n paste from the meter code ...)

[image]

georgweiss, 2023-04-11:
We are currently working on a new widget that might overlap with the proposed changes. For lack of a better word we have dubbed it "linear meter" (to distinguish it from the current "curved" meter).

Attached screen shot should give an idea on its features. Colors would be customizable, of course. 

This is work in progress, but when done we will make it publicly available as an optional widget that may be added to a site specific build of Phoebus.

[image]