ControlSystemStudio/phoebus #2657: Reading and writing to a single array element of a PVAccess PV

bsobhani, 2023-04-27:
The tooltip when typing in a PVAccess PV name says "pva://pva/subfield/subelement". If the PV is an array, is it possible to display or write to a single element? For example, if the array is "1 3 5 6 7", just display the 2nd element (3)?

kasemir, 2023-04-27:
For reading, yes, pva://array_pv[5] will display just that element.
But that only works at the end, pva://pva/subfield/subelement[5] is OK, pva://pva/subfield[4]/subelement is not.

It's still reading the whole array, we can't read just one element, so this is only syntactical sugar on the display end.
For related reasons, writing is not supported, since we can't just write pva://array_pv[5], we'd have to write the whole array.

bsobhani, 2023-04-27:
I see. I mainly wanted this for writing so if writing is not possible I have no use for displaying. I understand that it is a limitation of EPICS that writing to an array element is not possible.

For displaying, I will mention that pva://array_pv[5] did not work for me with PVs I generated with PvaPy (I tried both pva://array_pv[5] and pva://array_pv/value[5] but in the first case it just displayed the entire array, and in the second case it could not connect). But this is purely for your information since I do not need this feature.

closed 2023-04-27 by bsobhani