ControlSystemStudio/phoebus #2659: NTNDArray Decoding Failure

jeinstei, 2023-05-01:
I'm not sure where to start with this, as it looks like this could just be user error as much as an issue with NTNDArray decoding of a single-axis waveform. We have a waveform of RF data coming back to us in an NTNDArray, but the decoding seems to fail. 

The error below is created in the terminal when attempting to connect using the PV List in the most recent head on master (d8ed8062) built in my local system and passing all the tests. I'm probably on an "unsupported" java version (openjdk 20.0.1), but I also can't see how that could be causing this issue.

Happy to debug further if there is a debug guide I should be looking at for phoebus to enable some more details or we can modify the data structure being sent back.

```
2023-05-01 15:31:57 WARNING [org.phoebus.pv] Cannot decode 'my_ioc:Root:PvProcessor:AdcI[0]' [CID 2, SID 117768961 CONNECTED] = epics:nt/NTNDArray:1.0 
    union value
        short[] shortValue [4, -4, 5, 5, 2, 2, 8, 1, 1, 10, 5, 1, 3, 0, -4, 0, -9, 3, -2, -2, -5, 5, -4, 1, -4, -10, 1, -3, 0, -4, 2, 5, -2, -2, -8, 0, 9, -2, 5, -12, 4, -10, -1, -5, 9, 4, -6, -5, -7, 1, -1, 6, 1, 4, 4, 0, 4, -7, -6, -2, 1, 3, -5, 2, -3, 2, -9, -8, 1, -5, 10, 9, 1, -3, -4, -4, 0, 4, 5, -4, 5, -4, -3, 2, -2, 5, -2, 4, 2, -2, -2, 8, -6, 2, -9, 2, 5, -1, 4, -9, 2, 1, -1, 0, -1, 4, 4, 0, 9, 7, 0, 5, 2, -2, 2, 2, 6, -3, -1, 8, -1, -1, -9, 4, -3, 2, 2, 3, -5, 8, 1, -7, -2, -1, -6, 4, 4, 4, -2, -8, 4, -3, -7, -15, -10, -6, -10, -2, 5, 6, 3, 10, 2, 6, 1, 8, -1, -3, 3, 0, 0, 4, 4, 0, -5, 8, 0, 0, -6, 3, -3, 8, 5, 2, 1, 2, -5, -6, -1, -5, -9, -7, 8, 4, 7, -1, 5, -3, 3, 15, 2, -1, 0, 6, -2, 8, -9, 5, 2, 2, 0, 4, 13, -2, -11, 3, -5, -4, 1, 3, -5, 3, -4, -5, -2, -3, 3, 6, -2, -2, -1, 1, -3, 6, -2, -2, -6, -6, -10, -1, 3, 1, 1, -4, -4, -1, 0, 8, -4, 6, 0, 0, 0, 2, -9, 0, -3, 8, -1, 5, 3, 5, -2, -10, -1, -4, ...]
    codec_t codec
        string name null
        any parameters
            (none)
    long compressedSize 0
    long uncompressedSize 0
    int uniqueId 0
    time_t dataTimeStamp
        long secondsPastEpoch 0
        int nanoseconds 0
        int userTag 0
    alarm_t alarm [NO_ALARM]
        int severity 0
        int status 0
        string message null
    time_t timeStamp [1969-12-31 18:00]
        long secondsPastEpoch 0
        int nanoseconds 0
        int userTag 0
    dimension_t[] dimension
    epics:nt/NTAttribute:1.0[] attribute
java.lang.NullPointerException: Cannot invoke "String.isBlank()" because the return value of "org.epics.pva.data.PVAString.get()" is null
	at org.phoebus.pv.pva.ImageDecoder.decode(ImageDecoder.java:205)
	at org.phoebus.pv.pva.PVAStructureHelper.getVType(PVAStructureHelper.java:107)
	at org.phoebus.pv.pva.PVA_PV.handleMonitor(PVA_PV.java:88)
	at org.epics.pva.client.MonitorRequest.decodeValueUpdate(MonitorRequest.java:236)
	at org.epics.pva.client.MonitorRequest.handleResponse(MonitorRequest.java:198)
	at org.epics.pva.client.MonitorHandler.handleCommand(MonitorHandler.java:40)
	at org.epics.pva.client.MonitorHandler.handleCommand(MonitorHandler.java:21)
	at org.epics.pva.common.CommandHandlers.handleCommand(CommandHandlers.java:66)
	at org.epics.pva.client.ClientTCPHandler.handleApplicationMessage(ClientTCPHandler.java:331)
	at org.epics.pva.common.TCPHandler.handleMessage(TCPHandler.java:394)
	at org.epics.pva.common.TCPHandler.receiver(TCPHandler.java:294)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:317)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
	at java.base/java.lang.Thread.run(Thread.java:1623)
```

kasemir, 2023-05-02:
How do you generate this NTNDArray? From the area detector? We tried that with and without compression and never got into this situation...

The error you see is this:

```
java.lang.NullPointerException: Cannot invoke "String.isBlank()" because the return value of "org.epics.pva.data.PVAString.get()" is null
	at org.phoebus.pv.pva.ImageDecoder.decode(ImageDecoder.java:205)
```

That line of code is this:
```
if (name != null  &&  !name.get().isBlank())
```

Try changing that to this:
```
if (name != null   &&   name.get() != null  &&   !name.get().isBlank())
```

jeinstei, 2023-05-02:
Thanks @kasemir . This is from some code out of [slaclab/rogue](https://github.com/slaclab/rogue/blob/1a8b62db9205e38173c792fa42d52298ec4b97a1/python/pyrogue/protocols/epicsV4.py#L195) that we are interfacing with. Figuring out the "proper" way to handle such data so everyone can handle it would be the best, I'd think. I'm thinking we need to add support for the ScalarArray types.

I'll update the java for now and see if there's any other issues.

kasemir, 2023-05-03:
For arbitrary "custom" structures you'll typically also need a custom client, maybe in python.
For the NTNDArray, we only tested with those generated by the area detector.
But if what you generate from some python script is close enough to an area detector NTNDArray except for the codec being null, create a pull request for the above change and we'll include that.

So now that we get the NTNDArray into a "VImage", you can show a "VImage" in the image widget, but that only makes sense for actual images, not a 4096x1 array.
You can also show them in the XYPlot as a "Y" waveform.

Or use pva://that_name/value to just get the waveform

jeinstei, 2023-05-03:
Will get a short PR up shortly, then. You'd think it would show up in the XYPlot, but it didn't seem to -- I was going to take a closer look at that, and fortunately now I have a reason to.

kasemir, 2023-05-03:
The magic for plotting an "Image" as a Y (or X) in the XYPlot should simply be https://github.com/ControlSystemStudio/phoebus/blob/54c3eecda87dfe70885879ea73af2a5b3f862db7/app/display/representation-javafx/src/main/java/org/csstudio/display/builder/representation/javafx/widgets/plots/XYPlotRepresentation.java#L296

But might need to debug into that because something in your NTNDArray might look just different enough to not run through there.