ControlSystemStudio/phoebus #2662: Two issues in macro expansion

abrahamwolk, 2023-05-04:
I think there are two issues with the macro expanding phase.

**Issue 1**

When issuing a re-load of an OPI, the macro values of the OPI are not re-initialized to just the top-level macro-variable definitions of the OPI. Instead, the macro-variables defined at the top-level of the OPI are merged with the existing macro-variables of the running instance of the OPI. In case of naming conflict, top-level macro-variables overwrite existing macro values.This behavior makes it difficult to reason about the effect on macro-variables when issuing a "re-load" command to an OPI.

The current implementation is located in the function `loadModel()` in the class `DisplayRuntimeInstance`:
https://github.com/ControlSystemStudio/phoebus/blob/54c3eecda87dfe70885879ea73af2a5b3f862db7/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/app/DisplayRuntimeInstance.java#L360-L386

**Issue 2**
Consider the following function `expandValue()` in the class `Macros`:
https://github.com/ControlSystemStudio/phoebus/blob/54c3eecda87dfe70885879ea73af2a5b3f862db7/core/framework/src/main/java/org/phoebus/framework/macros/Macros.java#L272-L288

On line 284 it is checked whether the expanded form of the contents of a macro-variable is equal to the non-expanded form. Only if it is not equal to the expanded form is the binding of the macro variable updated (line 285).

Suppose the argument `input` of the function `expandValue` already contains a binding `B` for a macro-variable `$(M)`. Suppose further that the Java-variable `macros` contains a binding of `$(M)` to a literal value `V` with no free variables (i.e., no substitution will occur in the call to `MacroHandler.replace()`. After `expandValue()` has returned, `$(M)` is still bound to `B`, and not `V` which it, I believe, should be bound to.

**An example of how issue 1 and issue 2 cause a problem**
Counter-intuitive/wrong behavior may be observed in the following case (this is written using pseudo-code, not the actual syntax for macro assignments; `+` denotes string concatenation):

 - The OPI itself contains no top-level macro definitions.
 - A sub-OPI that is loaded when the main OPI is loaded contains two declarations of the form (this code is only loaded once when the OPI starts):
    - `$(A) = "SOMETHING`
    - `$(B) = $(A) + ":POSTFIX"`.
    
   (So that at this stage, it `$(B)` is set to `SOMETHING:POSTFIX`)
    
 - An button-press that can be initiated in the OPI sets `$(A) = $(B)`
 - When issuing a "re-load" of the OPI, both `$(A)` and `$(B)` have values assigned already: both have the value `SOMETHING:POSTFIX`. Since they are not defined on the top-level, the values are kept during re-load due to issue 1. When the sub-OPI that contains the declarations of `$(A)` and `$(B)` is loaded as part of loading the OPI, the result is that:
    - `$(A) == "SOMETHING:POSTFIX"` (due to issue 1 followed by issue 2, `$(A) is not updated)
    - `$(B) == "SOMETHING:POSTFIX:POSTFIX"` (due to `$(A)` having the wrong value, `$(B)` now has a wrong value)
    
However, the intention of the OPI designer is that `$(B) == "SOMETHING:POSTFIX"` instead of `$(B) == "SOMETHING:POSTFIX:POSTFIX"`.

**Proposed changes**
I propose two changes (fixing issue 1 and issue 2, respectively). The main motivation for the proposed changes is that they will make it easier to reason about macros in an OPI when issuing a "re-load" command to the OPI:

1. That the implementation of `loadModel()` in `DisplayRuntimeInstance` be changed so that the macro-variables are properly re-initialized when issuing a re-load command. (I.e., that all old macro values be discarded.)
2. That the test `if (! expanded.equals(orig))` on line 284 of Macros.java be removed. I.e., that `macros.put(name, expanded)` _always_ be invoked after macro expanding a macro level definition.

kasemir, 2023-05-04:
We generally don't support `$(A) = $(B)`, see https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility#macros, and changing anything about the macro handling risks breaking operational displays. Still, at first glance your proposed changes should be fine.

abrahamwolk, 2023-05-04:
Thanks for the link to the documentation describing the macro functionality in Phoebus. Is there available a more formal description of how it should work, or should I consult the implementation itself for that?

I agree that changing how macros work risks breaking currently working OPIs. On the other hand, the behaviors described in issue 1 and issue 2 can also result in bugs in OPIs, due to them being counter-intuitive and not-expected to the OPI designer.

abrahamwolk, 2023-05-05:
I have created a merge request with the changes described under "Proposed changes": merge request https://github.com/ControlSystemStudio/phoebus/issues/2662

I am not very familiar with the macro implementation and with how macros are "supposed" to behave in Phoebus, and therefore I cannot predict very well whether the changes can have adverse effects for existing OPIs or other unintended side-effects. I created the merge-request to have something concrete to discuss, as well as to facilitate testing.

kasemir, 2023-05-05:
For Issue 1: ```When issuing a re-load of an OPI, the macro values of the OPI are not re-initialized to just the top-level macro-variable definitions of the OPI. Instead, the macro-variables defined at the top-level of the OPI are merged with the existing macro-variables of the running instance of the OPI. In case of naming conflict, top-level macro-variables overwrite existing macro values.This behavior makes it difficult to reason about the effect on macro-variables when issuing a "re-load" command to an OPI.```

The state of a display is not only defined by the  just the top-level macro-variable definitions of the OPI. A display can be opened from an "Open Display" action that adds macros.
For example, a "Motor" display that uses "$(P)" for the motor prefix. It can be used as an embedded display, but it can also be used from an "Open Display" action that defines "P=MyMotor47". Either way, yes, if you define P in the display itself, that takes precedence over passed-in macros. So don't do that. The macro priority is described in training documents, see https://controlssoftware.sns.ornl.gov/training/2022_USPAS/Presentations/05%20Display%20Builder.pdf

There is the use case where a display is meant to be used by passing P into the display, but for testing you want it to "just work" without passing anything in. For that, you can use "$(P=default_value)" inside the display, but better would be creating a top-level "test" display that embeds or opens the display to be tested and passed in what's needed.

So Issue 1 looks like it's exactly as designed, sorry.

abrahamwolk, 2023-05-08:
Thanks for your answers! What I meant by a button-press setting $(A)=$(B) is that this is given as input to the original OPI; the actual working of the OPI where the unexpected behavior occurs is that one navigates from the original OPI to another OPI using a button-press, and then one navigates back again. After navigating back to the original OPI again, $(A)=$(B) has been set.

@kasemir wrote:
> So Issue 1 looks like it's exactly as designed, sorry.

In the case of issuing a "re-load" command to an OPI, wouldn't the expected behavior be that macro variables are re-initialized "from scratch"? I think that re-loading of the entire CS-Studio application isn't the same as issuing a "re-load" command to an OPI.

kasemir, 2023-05-08:
>  unexpected behavior occurs is that one navigates from the original OPI to another OPI using a button-press, and then one navigates back again, again using a button-press. 

"Back" and "Back using a button-press" are different.

1) Example displays, Macros, More Macro Hierarchy Examples
2) Open context menu on display, 'Macro Hierarchy' Information, and find "MY_MACRO=.. as defined in display"
3) Clock "Change MY MACRO", and in there you'll find "MY_MACRO=Set by action"

If you now use the "Back" button in the toolbar, you'll be back at step 2, with "MY_MACRO=.. as defined in display" etc.

On the other hand, if you add a button to the step 3 display that loads the same display as step 2, then that will be loaded with whatever context there was in step 3 plus macros set in the action. If you add "XXX=xxx" to the action, you'll see that when you navigate to step 2 via that button, but you won't see XXX defined if you use the "Back" button in the toolbar. If you try to set MY_MACRO, note that the display opened in step to defines that directly in the display, so whatever you pass into examples/macros/macros.bob for MY_DISPLAY will always be replaced by ".. as defined in display" since the macros.bob sets it to that.

> In the case of issuing a "re-load" command to an OPI, wouldn't the expected behavior be that macro variables are re-initialized "from scratch"?

No.

1) Open some main display
2) Open some motor.bob that needs MOTOR=motor1

When you check the Info of motor.bob, you'll find that it's something like motor.bob?MOTOR=motor1

So when you "Re-load Display" from the context menu or from a saved layout, you again get the motor.bob with MOTOR=motor1. Starting "from scratch" by just loading motor.bob without any MOTOR definition would mean you're not restoring the display, you see $(MOTOR):... instead of the actual PVs.

abrahamwolk, 2023-05-09:
Thank you for the detailed explanations. I conclude that I have misdiagnosed the issue, but I still believe there is a potential issue.

If we consider the following lines of code in the function `openDisplay()` in the class `ActionUtil`:
https://github.com/ControlSystemStudio/phoebus/blob/748ca07aab4fa45cf9b16ff238fe613d02069bec/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/ActionUtil.java#L104-L107
The intent of the quoted code is clearly that (top-level) macro bindings defined in the OPI that is loaded override (the second argument to `Macros.merge()` takes priority over the first) any macro bindings that are "inherited" from the "source OPI" that initiated the call to `loadDisplay()`. However, this only seems to be the case for top-level macro bindings of the OPI that is loaded, and not for macro-bindings in widgets that are part of the OPI that is loaded.

Suppose we have two OPIs, "A" and "B". A contains a button that opens B, and B contains a button that opens A. Upon first being loaded, A defines the macro binding `$(M) = "1"`. When B is loaded by pressing the button in A, `$(M) = "2"` is set. The question is what will happen when the button in B is clicked that loads A. It seems to depend on whether the `$(M) = 1` is defined (case 1) on the top-level of A, or (case 2) in a widget of A:

- **Case 1: `$(M) = 1` is defined on the top-level of A.** In this case `$(M) = "1"` after the call to `loadDisplay()`.
- **Case 2: `$(M) = 1` is defined in a widget of A.** In this case `$(M) = "2"` after the call to `loadDisplay()`.

That there is a difference in the value of `$(M)` in the two cases seems incongruous to me; I believe that case 1 and case 2 both should result in `$(M) = "1"` after the call to `loadDisplay()`.

kasemir, 2023-05-09:
What do you mean by "on the top-level of A"?

Yes, macros do to some extend reflect how a display was called.
Assume A.bob is called from someplace else with M=1. Then it opens B.bob, which in turn contains a button to open A.bob with M=2. So depending on how you call it, A.bob runs with M=1 or M=2. That's how we are able to create one "motor.opi" and then re-use that with various macro settings. It's not a bug but a feature.

abrahamwolk, 2023-05-10:
@kasemir wrote:
> What do you mean by "on the top-level of A"?

Suppose A contains a label widget. Then:
- By "on the top-level of A" I mean that the declaration `$(A)=1` is defined by first left-clicking on the background of the OPI (i.e., no widget is selected), and then clicking on the "Macros" button in the menu on the right (the "Properties" menu), and entering the definition there.
- By "defined in a widget of A" I mean that the declaration `$(A)=1` is defined by first left-clicking on the label widget (so that the label widget is selected), and then clicking  on the "Macros" button in the menu on the right, and entering the definition there.

When A is loaded using `loadDisplay()` from `B`, the value that `$(A)` ends up being bound to after A has been loaded is different in the two cases. My guess is that the second case (where the macro is defined in the label widget) should behave the same as the first case.

abrahamwolk, 2023-05-15:
After experimenting further, it seems that the problem also does not lie in macros defined in widgets, but rather with the hierarchical expansion of macros: counter-intuitive behavior is seen when widgets are defined in a Group-widget that contains a collection of widgets.

I have attached a small example that illustrates the counter-intuitive behavior.

The example is run by loading `macroIssue_A.bob` into Phoebus, and running it, and the performing the following actions:

1. Click on the button "Open B"
2. Click on the button "Open A"
3. Right-click and click on "Re-load Display".

Then:
 - After step 2 but before step 3, both `M3` and `M4` are equal to `1:POSTFIX`.
 - After step 3,  `M3` is equal to `2:POSTFIX`, while `M4` is still equal to `1:POSTFIX`.

Is this really the intended behavior?

For understanding the example: `macroIssue_A.bob` defines:

- The macros `M1` and `M4` on the "top-level".
- The macros `M2` and `M3` in the Group-widget.
- The action button "Open B" sets both `M1` and `M2` to `2`.

`macroIssue_B.bob` doesn't define any macros.
[example.zip](https://github.com/ControlSystemStudio/phoebus/files/11476935/example.zip)

kasemir, 2023-05-15:
Open macroIssue_A.bob
---------------------

Display file sets and displays M=1, M4=$(M1):POSTFIX
Group in there sets and displays M2=1, M3=$(M2):POSTFIX

=> M1=1, M4=$(M1):POSTFIX=1:POSTFIX are available anywhere in the display,
   M2=1, M3=$(M2):POSTFIX=1:POSTFIX are available inside the group.

Click on the button "Open B"
----------------------------

Button "Open B", which is outside the group, sets M1=2, M2=2 and opens macroIssue_B.bob

The macroIssue_B.bob display receives  M1=2, M2=2, M4=$(M1):POSTFIX, doesn't change anything, and displays M1, M2

=> M1=2, M2=2, M4=$(M1):POSTFIX

Click on the button "Open A"
----------------------------

Button "Open a" opens macroIssue_A.bob

macroIssue_A.bob is opened with the state seen by the "Open A" button, that is  M1=2, M2=2, M4=$(M1):POSTFIX
macroIssue_A.bob then sets M1=1,  M4=$(M1):POSTFIX, and the group inside the display sets M2=1, M3=$(M2):POSTFIX

It displays M1=1, M4=$(M1):POSTFIX=1:POSTFIX,
and the group in there displays M2=1, M3=$(M2):POSTFIX=1:POSTFIX

The 'info' for the display holds
 M1=1 (received as 2 then set to 1 in the display),
 M2=2 (received as 2. Set by group to 1, but outside the group it's 2),
 M4=$(M1):POSTFIX

Right-click to "Re-load Display"
--------------------------------

macroIssue_A.bob is opened with info M1=1, M2=2, M4=$(M1):POSTFIX

Display then sets  M=1, M4=$(M1):POSTFIX
Group in there sets and displays M2=1, M3=$(M2):POSTFIX

... but as you found the M3=$(M2):POSTFIX gets evaluated with M2=2, before the group sets it to M2=1.

Bummer
--------

When you replace DisplayMacroExpander.expandDisplayMacros with a NOP, it works more intuitively.

How we got there:
In the original design, we decided to keep macros as a map and expand them as late as possible.
These go together, since a map is generally not ordered.
If you define "A=a, AA=$(A)$(A)" and then receive them as a map in order "AA=$(A)$(A), A=a", that's OK as long as you evaluate as late as possible. If you try to evaluate "AA=$(A)$(A)" right away before "A=a" is known, you end up with an unresolved "$(A)".

This works fine for 99% of all displays where lower level displays are defined for re-use with macros, and higher level displays then call them with values for those macros.
But then there's the 1% of displays, typically converted from EDM, which contain
```
SAVE=$(A)
A=xxx
```
to invoke some sub-displays
and later try to restore `A=$(SAVE)`, which would give a recursive `A=$(SAVE), SAVE=$(A)  ==>  A=$(A)`.

This loop was broken by expanding some macros at load time, and that now causes the side effect that you see.

I see two options:

1) This is an esoteric example. It hasn't resulted in operational issues. Nobody's been unable to run their machine because of this. Touching the macro handling, on the other hand, could impact operational setups

2) The original design with map and late evaluation was wrong. We need a list of macros (or ordered map), and macros need to be expanded ASAP. `A=a` -> `Macros [ A=a ]`. `AA=$(A)$(A)` -> `Macros [ A=a, AA=aa ]`. If that breaks existing setups, we say sorry but it must be that way, that's how environment variables work, that's the only way to properly do it.

abrahamwolk, 2023-05-16:
Thank you for taking the time to work through the example and for writing the detailed explanation.

I started investigating this because of a bug report by one of our operators. An OPI he had implemented didn't work after the corresponding sequence of actions. In the OPI in question, there are several groups of widgets in the OPI, where macros are defined.

@kasemir wrote:
> Touching the macro handling, on the other hand, could impact operational setups

I agree that we must be very careful if we introduce any changes in how macros work. However, I think that this looks like a bug: the bindings in a Group widget should be evaluated in the right context.

The macros on the "top-level" of the OPI is evaluated in "its own" context:
https://github.com/ControlSystemStudio/phoebus/blob/28395307608fc6d9a1d99282e08513bac25567d5/app/display/model/src/main/java/org/csstudio/display/builder/model/macros/DisplayMacroExpander.java#L29-L32

However, this is not the case for macros defined "in widgets": they are evaluated in their parent's context (lines 40 and 44):
https://github.com/ControlSystemStudio/phoebus/blob/28395307608fc6d9a1d99282e08513bac25567d5/app/display/model/src/main/java/org/csstudio/display/builder/model/macros/DisplayMacroExpander.java#L34-L46

Shouldn't they, for consistency, also be evaluated in "their own" context? Under the assumption that `widget.getMacrosOrProperties()` returns the environment where the `widget`'s macros have extended the widget's parent's macro environment, I propose to change line 40 from

>`macros.get().getValue().expandValues(input);`

to

>`macros.get().getValue().expandValues(widget.getMacrosOrProperties());`

(If this indeed is a fix, then other place's where macros are expanded may also have to be modified accordingly; this code-path is only hit when I click "Re-load Display", not, for instance, when I navigate using action buttons. On the other hand, I only see the issue when I click "Re-load Display".)

kasemir, 2023-05-16:
I'd need some time to concentrate on this, which I don't have this week. But my gut feeling is that the late expansion was a bad choice. Instead of adding another tweak, the end result might be simpler, cleaner and predictable by replacing the macro handling with an ordered map and early expansion. Just need to find a volunteer who then handles the emails from users who are surprised by the change. What's your email address? Maybe redirect to chatgpt?

abrahamwolk, 2023-05-17:
I don't think that the proposed change is a tweak; it is intended as a (possible) bug fix. However, I am not 100% sure of this, since I am not very familiar with either the intended meaning of macros or their implementation.

My analysis is that it's not the late expansion that causes the problem in this case: upon action 3 in my example (clicking 'Re-load Display'), `DisplayMacroExpander.expand()` expands the macros of the Group widget in the wrong context: the context of the top-level where `M2 == 2`. The following is a screenshot of a debugger showing the context that is given as a parameter to `expandValues()` on line 40 of `DisplayMacroExpander.expand()` when the macros of the Group widget are expanded:

[image]

I suggested that, instead, the macros of the Group widget should be expanded in the context that is the result of extending the top-level's context by the macro definitions of the Group widget. The resulting context when applying my suggested change above is:

[image]

Under this change M3 indeed evaluates to `1:POSTFIX`.

Perhaps this way of extending the environment is wrong, too, though: it is possible that the bindings should be expanded and added to the context one-by-one, or that perhaps the intended semantics is something different again. As I wrote above, I am not familiar enough with the intended semantics to tell.

@kasemir wrote:
>But my gut feeling is that the late expansion was a bad choice. Instead of adding another tweak, the end result might be simpler, cleaner and predictable by replacing the macro handling with an ordered map and early expansion.

I agree that late expansion isn't a good idea in the presence of mutable state (variables can be updated). It is inherently difficult to reason about: it very easily becomes practically speaking non-deterministic since whether a macro is expanded before or after a variable is mutated will alter the value it is expanded to. Changing this now, however, may, as you say, break existing OPIs.

_If_ this is to be changed, I suggest investigating whether it would be preferable to introduce strict evaluation on a case-by-case basis, by introducing a new syntactic indication that certain variables should be expanded strictly (i.e., when they are defined). Using pseudo-code, then:

- The current syntax `M = <definition>` would denote the current behavior (leaving the meaning of existing macros intact)
- A new syntax, e.g., `M != <definition2>`, would denote strict expansion of `<definition2>` when `M` is defined.

I do not know whether the two expansion strategies can co-exist. However, if this approach is viable, then it would allow for strict evaluation on a case-by-case basis without breaking existing OPIs. (Perhaps late expansion could even be phased out in the long run.)

kasemir, 2023-05-19:
> DisplayMacroExpander.expand() ...
>  I propose to change line 40 from

    macros.get().getValue().expandValues(input);

> to

    macros.get().getValue().expandValues(widget.getMacrosOrProperties());

Yes, that looks good. Compared to #2665, the example from "Macros", "More Macro Hierarchy Examples", "Change MY MACRO" still functions with that change, plus it solves your `macroIssue_[AB].bob` example.

Do you want to create a pull request for that, it's certainly an improvement.

Eventually, we might still need to separate the macro spec from the runtime macro values, i.e., expanding all runtime values sooner rather than later. In the end, I think I want to go back to the behavior we have in EDM, MEDM, BOY, ... where you can basically configure this in the macro specs:
```
A=a
B=b
X=$(A)
A=$(B)
B=$(X)
```
.. and you then end up with values
```
A=b
B=a
X=a
```
instead of creating a loop `B=$(X)=$(A)=$(B)`

abrahamwolk, 2023-05-22:
I have created a merge-request and added you as a reviewer. One test broke, which I fixed. As I wrote above, I am not 100% sure of this change; I modified the comment of the test so that it reflects my reasoning.

**Regarding the late expansion:** after more thinking, I now think that there are two issues present that make it difficult to reason about the late expansion:
1. The presence of mutable variables. (As discussed above.)
2. The scoping is not lexical: a macro is expanded as far as possible (this is at least how I understand it; I am not sure of this) in the context that happens to be the current context when the expansion is triggered. Since it is in general not easy to predict in which context a macro will be expanded in, the consequence is that it can be very difficult to predict what the macro will expand to. (The "early macro expansion" code seems to implement lexical scoping, but I believe that not all macros are evaluated this way.)

Implementing lexical scoping would probably be a good idea from a macro language design point-of-view, since it would facilitate both understanding and debugging of macros. However, it would (in the presence of mutable variables) necessitate that a distinction be made between (A) variable _declarations_ and (B) _assignments_ to variables.

One option would then be (in order to avoid having to introduce this distinction) to simply disallow assignments to variables. Instead of assigning values to existing variables, one could declare a new variable with the same name that temporarily _shadows_ the binding of the same name of the surrounding scope. I can think of one use-case that cannot be accommodated in this way: one cannot _return_ values from an inner scope to its outer scope in this way; in order to use the environment as a return-value mechanism, one would need assignments to variables. On the other hand, return-values could perhaps be passed as parameter values to an OPI instead.

kasemir, 2023-05-22:
Early expansion is the way to go, so it's just like environment variables. Will take a little time, but started to look into it.

As for "one cannot return values from an inner scope to its outer scope in this way", that's true. A shell script cannot return data via environment variables, either. It receives them, can change them inside, but can't return:

```
X=1

bash
echo $X   -> 1
X=2
echo $X   -> 2
exit

echo $X   -> 1
```

For passing data not just into displays but also back out, you need to use PVs.

closed 2023-05-24 by abrahamwolk