ControlSystemStudio/phoebus #2672: Upgrade to Kafka 2.8 ( or 3.4?)

shroffk, 2023-05-15:
There have been recent improvements in kafka which improve error handling in streams.
The two changes which might be particularly useful are the improvements in how to handle timeouts and handling streams which enter error state

https://cwiki.apache.org/confluence/display/KAFKA/KIP-572%3A+Improve+timeouts+and+retries+in+Kafka+Streams

https://cwiki.apache.org/confluence/display/KAFKA/KIP-671%3A+Introduce+Kafka+Streams+Specific+Uncaught+Exception+Handler

georgweiss, 2023-05-15:
Thanks @shroffk.

tsgit, 2023-06-09:
yes, please update to kafka 3.4

currently cmd logging isn't working properly with phoebus alarm logger and kafka 3.4 
the logger doesn't create the  necessary consumers for both the message and cmd stream groups. it's one or the other

smarsching, 2024-07-19:
I think this has been addressed with #2953 and should be closed, right?

closed 2024-07-22 by shroffk