ControlSystemStudio/phoebus #2675: Newline char in PV name

georgweiss, 2023-05-16:
Not sure if this has been discussed...

If a PV name is copy-pasted into the PV name input field, the pasted string may contain a trailing newline (\n) character. This is often the case when copying strings from a web browser.

In Phoebus a trailing newline char is trimmed in ```https://github.com/ControlSystemStudio/phoebus/blob/master/core/pv/src/main/java/org/phoebus/pv/ca/JCA_PV.java``` L149,
but I cannot see the corresponding tweak in the pva code.

Any reason to not apply the same trimming in pva?

kasemir, 2023-05-16:
For the most part, we should pass PV names on because we don't know how the server side handles `xxx` vs `xxx.{dbnd:{abs:1.9}}`. But it should be OK to trim leading and trailing spaces as well as `[\r\n]`, because users enter PVs into probe etc and in there it's easy to have extra leading/training spaces and copied newlines.
Maybe instead of doing that inside the `JCA_PV` and the `PVA_PV` and the `Sim_PV` and the `Loc_PV` and the `MQTT_PV` we do that once inside `PVPool.getPV`?

georgweiss, 2023-05-16:
Thanks @kasemir. I'll dig a bit more and post a PR.