ControlSystemStudio/phoebus #2690: Spinner widget submits value twice

georgweiss, 2023-05-26:
When submitting a value using the ENTER key in the Spinner widget, the value is written twice. This was detected by an IOC developer seeing two pvput when using the widget.

Using a simple stand-alone JavaFX app I confirmed that the native JavaFX spinner actually does trigger two KEY_PRESSED events for the ENTER key, but not for other keys (e.g. ESC).

Have not found anything related to a bug report on this.

One change that seems to solve this issue is to change
```
 case ENTER:
      // Submit value, leave active state
      submit();
      setActive(false);
      break;
```

to

```
 case ENTER:
        // Submit value, leave active state
        if(active){
            submit();
            setActive(false);
        }
        break;
```

Any reasons not to solve it this way?

kasemir, 2023-05-30:
> .. value is written twice. This was detected by an IOC developer seeing two pvput when using the widget.

Fundamentally, that needs to be accepted on the IOC end.
Multiple operators in the control room might have the same display open and enter values at about the same time.
A "slider" writes values in rapid succession while it's moved.
Scripts or other automation tools can write many values.
I've seen how an operator, instead of simply clicking an "On" type of button, was click-click-click-clicking it, simply an odd habit. In that specific case, it was indeed a problem because each 'click' resulted in writing a message to a PLC. The issue was solved by setting the "HIGH" field of the associated bo record too something like 3 seconds.
So bottom line, the IOC should be robust against such writes.

.. but at the same time we should not cause extra unexpected  writes like these from a spinner.

>  I confirmed that the native JavaFX spinner actually does trigger two KEY_PRESSED events for the ENTER key

Good that you tracked it down to the JFX spinner itself.
Workaround looks reasonable.

georgweiss, 2023-05-30:
@kasemir, thanks for your feedback.

Adding the workaround would in my view make sense as there is a subtle difference: spinner up/down button will submit value once, while ENTER submits twice. Even if this should not be an issue on the IOC side, users should not be surprised to learn that the PV value was written twice.

closed 2023-05-31 by georgweiss