ControlSystemStudio/phoebus #2695: Translations are observed instead of rotations when rotating Picture or Symbol widgets

abrahamwolk, 2023-06-01:
The merge-request https://github.com/ControlSystemStudio/phoebus/pull/2685 can cause issues when Picture or Symbol widgets are rotated: instead of a _rotation_, a _translation_ of the image can be observed.

This issue has been observed on Mac, Linux, and Windows. Under Windows it has been observed to depend on the computer that runs Phoebus: in at least one instance the problem has _not_ been observed, and in at least one instance (on a different computer) the problem _has_ been observed. Under Windows, at least, it therefore seems to depend on the specific setup.

The cause is the setting in https://github.com/ControlSystemStudio/phoebus/pull/2685 of `imageView.setCacheHint(CacheHint.SCALE)`. Why this setting causes the problem is not understood. A guess is that instead of _optimizing_ for a scaling transformation, the graphics system in these cases continues under the assumption that _only_ scaling transformations will be applied.

Experimentation has shown that instead setting `imageView.setCacheHint(CacheHint.SPEED)` seems to work.

The fact that the original option (`CacheHint.SCALE`) didn't work on some setups, suggests that it is conceivable that `CacheHint.SPEED` may also fail on some setups. Such a failure has not been observed yet, but only very limited testing has been carried out.

The question is how to fix this. Options include:

 1. Change from `CacheHint.SCALE` to `CacheHint.SPEED`
 2. Add a boolean option to `Phoebus.ini` that when set to `false` turns off the caching (i.e, it leaves the caching as it was before https://github.com/ControlSystemStudio/phoebus/pull/2685), and when set to `true` enables `CacheHint.SPEED`. It is an open question what the default value of the option should be.
 3. Add an option to `Phoebus.ini` that allows for a selection between all available `CacheHint` options: `CacheHint.DEFAULT`, `CacheHint.SPEED`, `Cachehint.SCALE_AND_ROTATE`, `CacheHint.SCALE`, `CacheHint.ROTATE`, and `CacheHint.QUALITY`. (The different options are documented at [1].) The rationale for this option would be that since the behavior is outside of our control and we cannot possibly test all kinds of setups, we cannot rule out that any option we choose doesn't fail on some setup.  Under this option, the user can test the available options and choose the one that works on their setup. It is an open question what the default value of the option should be.
 
[1] https://docs.oracle.com/javase/8/javafx/api/javafx/scene/CacheHint.html

kasemir, 2023-06-01:
Since the behavior is somewhat unpredictable, a configuration option might be best.

This seems similar to the GTK2/3 issues on Linux, where `-Djdk.gtk.version=2` had been necessary to enable drag/drop. https://github.com/ControlSystemStudio/phoebus/wiki/JavaFX

Can you check if maybe there is already a similar option for the cache hint? Maybe `prism.cacheshapes` can already be used to configure the default cache behavior?
 http://werner.yellowcouch.org/log/javafx-8-command-line-options
 https://www.jfx-central.com/tips/application_flags

shroffk, 2023-06-01:
If we cannot find an existing command line option like suggested about, I would vote for option 3 ( I am assuming we are still limited to picture and symbol only )

I don't want to trigger Kay's vertigo again but I also expanded my test case to include 160 widgets, set -Dprism.maxvram=100M... and still did not succeed is recreating the issue on my windows machine :(

In this case, the default cache option is better than the SCALE option since the images render better. Also, by having a preference I can set it to different cache options for different OS deployment. This way we don't have to make decisions based on behaviour on one specific OS.

abrahamwolk, 2023-06-07:
I tried running with different command-line options that seemed relevant, but I was unfortunately not able to fix the issue that way.

I will implement option 3, with the default set to the default caching behavior.

closed 2023-06-08 by georgweiss