ControlSystemStudio/phoebus #2696: Can't connect to IOC using name_servers parameter

KacperKlys, 2023-06-01:
I have channel access 5064 port forwarded using SSH from remote to my local machine to port 45064. I am able to configure `caget` to read PVs from the port (by setting "EPICS_CA_NAME_SERVERS=" together with empty EPICS_CA_ADDR_LIST and "EPICS_CA_AUTO_ADDR_LIST=NO").

Is there any way to configure Phoebus to find PVs on port 45064? I tried to set `org.phoebus.pv.ca/name_servers=x.x.x.:45064`  in the Phoebus configuration file but it does not work. Am I missing something? I saw the old issue #988 so I assume this parameter work? Maybe I'm configuring something wrong?  How `org.phoebus.pv.ca/addr_list` and `org.phoebus.pv.ca/auto_addr_list` parameters should be set while setting `name_servers` ?

Cheers
Kacper

kasemir, 2023-06-01:
Verify that you're indeed setting it.
In the GUI, open Help, About, Show Details.
Under Preference Settings, check that `org.phoebus.pv.ca/*` is as expected, 
then under System Properties, check if `com.cosylab.epics.caj.CAJContext.*` has the expected copies.

KacperKlys, 2023-06-02:
That's what I always do to check if settings file was loaded correctly. So I am pretty sure that the Phoebus matches the one in the settings file.

KacperKlys, 2023-06-15:
Hi, I have followed your instructions and everything works well until IOCs and Phoebus instance are on the same host. When I ran the IOCs on one host and Phoebus on the other  I was unable to configure Phoebus to read "ramp2" (hosts are in the same network, it's my internal network, no firewall). Is this how it should work?

hydro-controls, 2024-05-02:
I know I am bumping an old thread, but I came across this nearly identical issue as well. When Phoebus is resolving PV names over an SSH tunnel, it attempts to connect to the actual server port (e.g., EPICS_CA_SERVER_PORT) set on the remote IOC instead of the endpoint on the local host. For example:
1. The IOC Server (10.10.10.10) is running a CA IOC on port 5066
2. The client machine (10.10.10.11) builds an SSH tunnel to the IOC Server:
```
$ ssh -L 9999:localhost:5066 tester@IOCServer
```
3.  Phoebus is started with:
```
# Disable default use of UDP broadcasts
org.phoebus.pv.ca/auto_addr_list=false
org.phoebus.pv.ca/addr_list=
# Configure only second IOC for TCP port name lookup
org.phoebus.pv.ca/name_servers=127.0.0.1:9999
```
4. Phoebus will fail to connect to a PV and issue an error that it cannot connect to localhost:5066.
5. Normal command line tools work as expected when using EPICS_CA_* environment variables.
6. If you assign the same port number to both ends of the SSH tunnel, Phoebus will retrieve data.

I didn't take the time to chase this down, but I expect it is an issue in the Java implementation of Channel Access: https://github.com/epics-base/jca

For whatever reason, Phoebus is resolving the PV IOC address to the actual port on the remote machine. My understanding is the C-implementation returns 0 instead of an address/port if a TCP request is made to a CA server port and the PV exists on that port. This may be why the the C-implementation functions correctly.

kasemir, 2024-05-03:
Certainly looks like a jca issue, but it's good tradition to shoot the messenger.

Still, to fix this on the JCA side,  the protocol basically works like this:

1) Client sends search request. Could be done via UDP broadcast, direct UDP message, or TCP.
2) A server that recognizes the channel name replies with "I have that channel, talk to me on TCP port XXX"

That works fine as long as the client directly reaches the server.
In your example, an ssh tunnel is added, so you could have this:

1) Client sends search request to ssh tunnel at localhost:9999
2) ssh tunnel forwards that search request to server at 10.10.10.10:5066
3) Server replies: I have that channel, talk to me at 10.10.10.10:5066

The server has no idea about the tunnel. It cannot say "talk to me at localhost:9999".

What we did with the PVAccess implementation, and what might also happen with the C implementation of channel access: If the search is received via TCP, the server replies with "talk to me on this already established TCP connection".

1) Client sends search request to ssh tunnel at localhost:9999
2) ssh tunnel forwards that search request to server at 10.10.10.10:5066
3) Server recognizes that the search was received via TCP, so it replies: I have that channel, talk to me within this TCP connection that we have already used for the search.

Technically, that's expressed as returning a zero address. The C implementation, which means the CA server on the IOC, might already do that, so maybe the missing piece here is that the JCA client needs to recognize "talk to me at 0.0.0.0:0" as "talk to me on the already established TCP connection"?

tynanford, 2024-05-03:
FWIW I am able to view PVs in phoebus from a channel access gateway via SSH tunnels with a jump host in between. And it doesn't depend on which local port I pick.

`ssh -fnNT -L 9999:ca-gateway-ip:5064 -J user@jump-host user@internal-client-machine`

But I see the same behavior as described above for tunneling connections to a single IOC (local/remote ports have to match)