ControlSystemStudio/phoebus #2705: Refactor of Marcos.expandValue

georgweiss, 2023-06-09:
Seems #2697 has refactored ```Macros.expandValue()``` to ```Macros.expand()```. Unfortunately this affects some of our OPIs where creative developers have peeked into the Java code.

Any reason to not maintain ```expandValue``` for backwards compatibility?

georgweiss, 2023-06-09:
> And again whoever writes scripts and has to ask for help shouldn't do it, and whoever still does write scripts must be prepared to maintain them..

Agree. But recommendations - no matter how strong and clear - have a tendency to be ignored.

georgweiss, 2023-06-09:
I may actually suggest rename back to ```Macros.expandValue```

kasemir, 2023-06-09:
OK, I'll do that.
That way, it'll look the same, there will be no immediate error from the scripts, but they might no longer behave as originally. That will let it sink in that those who write scripts need to maintain them, they may fail without any obvious error message.

kasemir, 2023-06-09:
..but it was `expandValues()`, plural, right?

georgweiss, 2023-06-09:
Not sure...

I have verified that skipping the call solves the problem for the case at hand.

georgweiss, 2023-06-09:
It's plural

kasemir, 2023-06-09:
It was `public void expandValues(final MacroValueProvider input) throws Exception`, now it'll be `public void expandValues(final Macros base)`.

So same name, but really needs `Macros` which can iterate the values to copy, plus then expand them, not just a value provider. So some scripts will still fail because they don't have complete Macros, and again simply skipping the call is the correct solution.

kasemir, 2023-06-09:
Try now that there is again an `expandValues` method. Do the scripts actually work when calling that?

closed 2023-06-09 by georgweiss