ControlSystemStudio/phoebus #2726: Tank widget log scale

ttratnik, 2023-06-27:
Hi,

I am having troubles with the log scale of the tank widget.

It doesn't seem to display correct values based on the minimum and maximum values of the scale.
In the example I am attaching the minimum of the widget is set to 1E-9, the maximum to 1E-4 and
limits from PV is turned off.

Best regards
Tomaž

[image]
[image]
[image]

kasemir, 2023-06-27:
@abrahamwolk  at ESS is in the middle of creating new bar/tank/scale type widgets, https://github.com/ControlSystemStudio/phoebus/discussions/2718, and overhauling the log axis support, https://github.com/ControlSystemStudio/phoebus/pull/2719, so maybe this can become part of that?
Or you wait for the linear meter widget and then switch to that?

ttratnik, 2023-06-27:
I decided that I am going to use the progress bar widget whose log scaling works but doesn't have the option to show the scale on the left side. I will just make the scale myself. Maybe I will switch to a different widget later.
Thank you for the reply

georgweiss, 2023-06-27:
FYI: #2718 does not resolve this issue.

@abrahamwolk is back from vacation Thu July 6th.

abrahamwolk, 2023-07-05:
@kasemir wrote:
>@abrahamwolk at ESS is in the middle of creating new bar/tank/scale type widgets, https://github.com/ControlSystemStudio/phoebus/discussions/2718, and overhauling the log axis support, https://github.com/ControlSystemStudio/phoebus/pull/2719, so maybe this can become part of that?
Or you wait for the linear meter widget and then switch to that?

Unfortunately, the work on these two items are unrelated: the changes to the logarithmic scale are not related to the Linear Meter, and the Linear Meter currently does not support a logarithmic scale.

mellguth2, 2023-10-27:
I have no experience with pull requests or may not be authorized, but I checked the following code change and it provided a sensible behaviour of the tank widget when configured with logarithmic scale.
Maybe it can help with fixing this issue.
```
git diff HEAD~1 HEAD
diff --git a/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/RTTank.java b/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/RTTank.java
index 90577ded8..826546c33 100644
--- a/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/RTTank.java
+++ b/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/RTTank.java
@@ -275,8 +275,19 @@ public class RTTank extends Canvas
             level = plot_bounds.height;
         else if (max == min)
             level = 0;
-        else
-            level = (int) (plot_bounds.height * (current - min) / (max - min) + 0.5);
+        else {
+            if (scale.isLogarithmic()) {
+                if (min <= 0 || max <= 0.0 || current <= 0.0) {
+                    level = 0; // refuse to try any mapping if negatives or zero are involved
+                }
+                else {
+                    level = (int) (plot_bounds.height * Math.log(current/min) / Math.log(max/min));
+                }
+            }
+            else { // linear scale
+                level = (int) (plot_bounds.height * (current - min) / (max - min) + 0.5);
+            }
+        }
 
         final int arc = Math.min(plot_bounds.width, plot_bounds.height) / 10;
         gc.setPaint(new GradientPaint(plot_bounds.x, 0, empty, plot_bounds.x+plot_bounds.width/2, 0, empty_shadow, true));

```

kasemir, 2024-02-02:
@mellguth2 thanks for your suggestion, works fine and has been incorporated only a few months later...

closed 2024-02-02 by kasemir