ControlSystemStudio/phoebus #2743: Change color title of group container widget independently

Mathis-Hu, 2023-07-24:
It would be practical to add the possibility of changing the title color of a group container independently of the background color and the color of the line/title bar, because sometimes for the title bar style, the title label is barely visible and we do not want to change all the background or bar color. I did a first version in the code, but I'm creating an issue first rather than creating a pull request to ask if the design is ok with the rest of phoebus.
Before :
[image]
[image]

After :
[image]
[image]

Actually, foreground property is for the line and bar color (and label if the style is not TITLE) and background is for the background color (and label if style is TITLE)
In my modification, I looked how it's done with other widget and I'm doing foreground for label (no matter the style), line color property for lines & bar color, and background is still for background color.
To do the conversion with older view version, I had increase the GroupWidget version from 2.0.0 to 3.0.0.

Mathis-Hu, 2023-07-24:
@shroffk

mcnanneyd, 2023-08-16:
It looks like this commit is what is breaking our BOY to Phoebus conversions. I really like the addition of an independent color title. However, it does seems to break almost all of our screens during conversion. Here is the opi screen before conversion:
[image]
And its attributes:
[image]

After converting to bob, the screen looks as follows:
[image]
And its attributes:
[image]

So, the border color should be mapped to line color, but it seems to be completely ignored. Is there a way to implement the conversion to meet all of our needs? Thanks!

kasemir, 2023-08-16:
This was somewhat expected, see comment in PR:
"It's possible that this conversion needs to be tweaked as we open a bunch of older displays and find that we missed something, but that would then be a follow-up update."

So now it's time for such a follow-up ;-)

Mathis-Hu, 2023-08-16:
I'm on it

Mathis-Hu, 2023-08-16:
Ok, I'd maked the changes, I'll make a PR.
When merged, you should have the same result as if you had opened it in Phoebus V4.7.2 @mcnanneyd 
Sorry for the inconvenience.

Also, while testing I noticed a difference between Phoebus & CSS (even in 4.7.2), in CSS, it seems that you cannot modify the color title at all and it's light gray (when border style is title bar), same for the line in group box style, it's taking a default darker color based on the background color. While in Phoebus, all colors depends on user choices, so you may have some difference in the conversion.
Is this behavior desirable ? @kasemir

kasemir, 2023-08-16:
> may have some difference in the conversion...  Is this behavior desirable?

Good question. I would prefer for the only group style options to be "Group Box" and "None", with color fixed to black. Very little to configure, all displays look consistent, and the conversion to the next display tool will be easier. But it's too late, we have the "Title" and "Line" options plus now all colors are configurable, so if somebody doesn't like the color, they can change it to whatever the next person won't like.

For the conversion from other display tools, first let's remember that BOY, the display builder, the web runtime are 3 quite different tools, with completely different graphics libraries.  The fact that you can simply open an older display file and it mostly works is already an accomplishment.

If you can tweak the conversion based on the version numbers, and have logic like "if older version, line_color = background_color.darken()", that's great. Maybe handle that as issues arise, just as you did in this case.

mcnanneyd, 2023-08-16:
> 

@Mathis-Hu , this is a nice feature that I am glad we have now. Thanks for the quick fix!

Mathis-Hu, 2023-08-16:
You're welcome, don't hesitate to report other issue if you find some :)

closed 2023-09-19 by shroffk