ControlSystemStudio/phoebus #2747: Choice widget behavior

nusaqib, 2023-07-25:
I recently built the Phoebus 4.7.3 and noticed that choice button abnormal behavior in terms of layout. The last item is in opposite orientation to the selected orientation (horizontal/vertical) as seen in the attached screenshots.
[image]
[image]

kasemir, 2023-07-25:
Haven't seen that. Looks OK for me:

[image]

nusaqib, 2023-07-25:
Okay, so this was strange. I was noticing same behavior on Rocky 9 and Debian 12 but both had MATE desktop environment. I switched to GNOME and now it's working as expected.

kasemir, 2023-07-25:
In your screenshot, I'm not sure the horiz and vert alignment is as desired, either.
XFCE seems to work OK, but not sure how the different desktops influence this.
For what it's worth, on Linux, the javafx graphics are based on GTK. For the time being you can select either GTK2 or GTK3, see https://github.com/ControlSystemStudio/phoebus/wiki/Graphics-Performance. Maybe that makes a difference?

nusaqib, 2023-07-25:
Thanks Kay! Running with GTK2 on MATE resolved this.

closed 2023-07-25 by nusaqib

kasemir, 2023-07-26:
Well, good that it works with GTK2. Unfortunately, the 'current' GTK version is 4. GTK2 support in JavaFX has been deprecated a little while ago. Not sure when it will be fully removed.

https://github.com/openjdk/jfx/blob/jfx20/doc-files/release-notes-18.md#deprecate-javafx-gtk-2-library-for-removal

https://bugs.openjdk.org/browse/JDK-8273089