ControlSystemStudio/phoebus #2764: build error: fail during UpdateThrottleTest

caraxlr, 2023-08-10:
Hi, I got some error when trying to run mvn clean install. mvn -DskipTests  clean install -e can work fine. here is the error shown:

[INFO] Running org.phoebus.ui.javafx.UpdateThrottleTest
-> Trigger..
<- Update!
-> Trigger..
-> Trigger..
-> Trigger..
<- Update!
<- Update!
<- Update!
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 2.821 s - in org.phoebus.ui.javafx.UpdateThrottleTest
[INFO] Running org.phoebus.ui.javafx.ImageCacheTest
Aug 10, 2023 1:37:26 PM com.sun.javafx.application.PlatformImpl startup
WARNING: Unsupported JavaFX configuration: classes were loaded from 'unnamed module @213a2e92'
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 1.138 s - in org.phoebus.ui.javafx.ImageCacheTest
[INFO] Running org.phoebus.ui.javafx.svg.SVGTranscoderTest

(java:460464): GLib-GObject-CRITICAL **: 13:37:28.960: cannot register existing type 'GSettings'

(java:460464): GLib-GObject-CRITICAL **: 13:37:28.960: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(java:460464): GLib-CRITICAL **: 13:37:28.960: g_once_init_leave: assertion 'result != 0' failed

(java:460464): GLib-GObject-CRITICAL **: 13:37:28.960: g_object_new_valist: assertion 'G_TYPE_IS_OBJECT (object_type)' failed

Thanks for your help!

kasemir, 2023-08-10:
UpdateThrottleTest completes, note `Tests run: 1, Failures: 0, Errors: 0`.

What fails is SVGTranscoderTest.

caraxlr, 2023-08-10:
edited, thanks!

kasemir, 2023-08-10:
Of course no idea why it fails. Georg might have an idea, @georgweiss 
Maybe tell us if this is on Linux, Windows, ..., real computer or a VM without graphics ...

kasemir, 2023-08-10:
GLib sounds like Linux...

caraxlr, 2023-08-10:
It's Linux, RHEL 8.8

georgweiss, 2023-08-14:
Unfortunately I have no way to test on RHEL 8.8. 

Here is a list of things you could try (in this order):

1. Run the build with GTK2, e.g. ```mvn -DjvmArgs="-Djdk.gtk.version=2" install```
2. Build using JDK17 or later
3. Update javafx version in top level pom.xml from 19 to 20 (requires JDK17).

georgweiss, 2023-08-15:
To me this is a new type of error. Would you be able to determine in which of my suggested steps this error was first shown?

The libawt_xawt.so shared library is part of the Java runtime, see screen shot below taken from a Java 11 installation. The path shown in your error seems a bit odd. Does the folder /home/phoebus3/XIAOL/anaconda3/envs/phoebus17/lib/jvm/lib even exist? If so, what does it contain?

I would maybe download an OpenJDK package (from https://adoptium.net/temurin/releases/) and unzip. Then update your JAVA_HOME environment variable to point to that JDK and try to rebuild.

[image]

caraxlr, 2023-08-15:
Hi Georg,

Here is the screenshot of 
/home/phoebus3/XIAOL/anaconda3/envs/phoebus17/lib/jvm/lib

[image]

It's true that there is no libawt_xawt.so file there. The error was shown when I tried using JDK17.

After I changed to the openJDK package listed in the given website, the build was successful. Thanks!

closed 2023-08-15 by caraxlr