ControlSystemStudio/phoebus #2770: phoebus screen no response

caraxlr, 2023-08-15:
Hi,

Sometimes in the middle of using Phoebus CSS, my computer screen would be frozen and I have to do a "ps -ef|grep gnome​" to find the "/usr/bin/gnome-shell" and kill it to get my screen back. I wonder if there is any possible reason? And is there any parameter to inspect the problem?

Thanks a lot!

caraxlr, 2023-08-15:
some people in my group also report that phoebus may be very slow sometimes. And I found this similar problem occurred before? https://epics.anl.gov/tech-talk/2021/msg02012.php

georgweiss, 2023-08-16:
> my computer screen would be frozen...

Do you run Phoebus on your computer or are you using a remote server/desktop?

kasemir, 2023-08-16:
On Linux, the JavaFX library uses GTK, and it tries to use the native graphics hardware for best performance. When logging in remotely, that can be counterproductive. For example, I've seen cases where the process used several CPU cores, but a profiler showed that the time was all spent inside the native graphics library and not in the java code.

See https://github.com/ControlSystemStudio/phoebus/wiki/JavaFX about adding `-Dprism.verbose=true` and `-Djdk.gtk.verbose=true` to your launcher script, so you can then see if the "Prism pipeline" uses "es2" or "sw", and if GTK uses GTK2 or GTK3.
Then experiment with `-Dprism.order=sw` to disable the  native graphics hardware. While es2 is best for a local session in front of the actual monitor/mouse/keyboard, remote sessions tend to run better without.

caraxlr, 2023-08-16:
yes, I'm running on local computer

caraxlr, 2023-08-16:
not sure what to check for this kind of performance issues.. I just ran into this freezing problem once, but my team members said it may freeze or be very slow. Thanks again!

georgweiss, 2023-08-17:
With factors like remote desktop issues eliminated...

Phoebus becoming unresponsive or very slow _could_ be a memory issue. As a rule of thumb, the default maximum amount of memory available for the JVM is 25% of the total RAM. You can however find out the exact number using ```java -XX:+PrintFlagsFinal -version | grep -iE 'HeapSize|PermSize|ThreadStackSize'```, look for ```MaxHeapSize```. 

The JVM option ```-Xmx<size>``` gives you more control over the maximum memory, but of course one must be careful not to give the JVM too much of the total system memory.

I hesitate to give exact recommendations on how much total RAM a computer would need to run Phoebus as it depends on the use case.

Examples of when my users have encountered unresponsiveness due to memory issues:

1. Scripts where PVs are created but not released. This causes memory leaks.
2. Very large data buffers in the Data Browser.

caraxlr, 2023-08-30:
I've run the command: java -XX:+PrintFlagsFinal -version | grep -iE 'HeapSize|PermSize|ThreadStackSize', the maxHeapSize is 7.7GB. Is that enough? I think the engineers are using it data-intensively. And The live_buffer_size is set as 5000. 

I think my group are using the same java and product-4.6.6-SNAPSHOT.jar to run phoebus, is that a possible cause of the problem?

"$TOP/../java" -Dfile.encoding=UTF-8 -jar $JAR $OPT "$@" &

Thanks again.

georgweiss, 2023-08-31:
7.7 GB is quite enough. The buffer size of 5000 is the default size, this should not be a problem unless you're displaying a very, very large number of data browser plots.

As for the actual Phoebus version... I recommend to always run the latest available, which at this point is 4.7.3-SNAPSHOT. If nothing else you get bug fixes and new features.

Again, memory leaks can be created in user-defined scripts when using the PV APIs.

You could consider to use a profiler like https://visualvm.github.io/ (free) or https://www.ej-technologies.com/products/jprofiler/overview.html (not free) to find potential memory leaks and/or CPU intensive code.

caraxlr, 2023-08-31:
we are using the java and SNAPSHOT from the same location, is it a possible cause? I will try to inspect whether we have the memory leaks, thanks for the info

closed 2023-09-28 by caraxlr