ControlSystemStudio/phoebus #2783: Sometimes the horizontal layout for choice buttons are not respected

Mathis-Hu, 2023-08-22:
When I create a choice button, with certain dimension, I the horizontal layout boolean is not respected.
[image]
And if I put it in Vertical :
[image]

If vertical, the problem seems to persist independently of the number of items, but if horizontal, it seems to do the problem only on odds number of items (except 2). 
With 3 items
[image]
With 4 items
[image]

@shroffk

kasemir, 2024-02-21:
While I cannot dependably reproduce this, I've now seen this happen when moving a display, be it runtime or editor, between monitors.
The layout is right now computed for an exact fit, so in the horizontal case the width of each button is `widget_width / number_of_buttons`. Looks like that layout is too tight, and JFX sometimes decides to spill the last button over into the next row or column.
Will simply make the buttons a little smaller to avoid this.

Mathis-Hu, 2024-03-25:
Thank you for the change

If this issue is resolved, can I close it ?

closed 2024-03-25 by kasemir