ControlSystemStudio/phoebus #2795: Fix errors on javadoc generation on core-formula

katysaintin, 2023-08-30:
Fix javadoc generation on core-formula 
Task defined in the codathon context
`mvn compile javadoc:javadoc`
generates following errors
`
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-javadoc-plugin:3.1.1:javadoc (default-cli) on project core-formula: An error has occurred in Javadoc report generation:
[ERROR] Exit code: 1 - D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\Formula.java:51: error: malformed HTML
[ERROR] *  <li>comparisons <, >, >=, <=, ==, !=
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\Formula.java:51: error: malformed HTML
[ERROR] *  <li>comparisons <, >, >=, <=, ==, !=
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\Formula.java:52: error: bad HTML entity
[ERROR] *  <li>boolean logic !, &, |,  .. ? .. : ..
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\Node.java:19: error: reference not found
[ERROR] *  @exception on error
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\Node.java:26: warning: no @param for node
[ERROR] public boolean hasSubnode(Node node);
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\Node.java:31: warning: no @param for name
[ERROR] public boolean hasSubnode(String name);
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:28: warning: no @param for name
[ERROR] public VariableNode(final String name)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:34: warning: no @param for name
[ERROR] public VariableNode(final String name, final double value)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:34: warning: no @param for value
[ERROR] public VariableNode(final String name, final double value)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:40: warning: no @param for name
[ERROR] public VariableNode(final String name, final VType value)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:40: warning: no @param for value
[ERROR] public VariableNode(final String name, final VType value)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:52: error: @param name not found
[ERROR] /** @param New value of variable. */
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:53: warning: no @param for value
[ERROR] public void setValue(final double value)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:58: error: @param name not found
[ERROR] /** @param New value of variable. */
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\VariableNode.java:59: warning: no @param for value
[ERROR] public void setValue(final VType value)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\array\ArrayRangeOfFunction.java:33: warning: no comment
[ERROR] public class ArrayRangeOfFunction extends BaseArrayFunction {
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\array\ScaleArrayFormulaFunction.java:65: warning: no description for @throws
[ERROR] * @throws Exception
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\math\Abs.java:4: warning: no comment
[ERROR] public class Abs extends OneArgMathFunction
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\math\Abs.java:6: warning: no comment
[ERROR] public Abs()
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\math\ACos.java:4: warning: no comment
[ERROR] public class ACos extends OneArgMathFunction
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\math\ACos.java:6: warning: no comment
[ERROR] public ACos()
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\math\ASin.java:4: warning: no comment
[... 167 lines cut ...]
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\NotEqualNode.java:17: warning: no comment
[ERROR] public NotEqualNode(final Node left, final Node right)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\NotNode.java:17: warning: no comment
[ERROR] public NotNode(final Node n)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\OrNode.java:17: warning: no comment
[ERROR] public OrNode(final Node left, final Node right)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\PwrNode.java:17: warning: no comment
[ERROR] public PwrNode(final Node left, final Node right)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\RndNode.java:17: warning: no comment
[ERROR] public RndNode(Node n)
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\SPIFuncNode.java:34: error: @param name not found
[ERROR] *  @param n Argument node
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\SPIFuncNode.java:36: warning: no @param for args
[ERROR] public SPIFuncNode(final FormulaFunction function, final Node args[])
[ERROR] ^
[ERROR] D:\Katy\Git\phoebus2\core\formula\src\main\java\org\csstudio\apputil\formula\node\SubNode.java:17: warning: no comment
[ERROR] public SubNode(Node left, Node right)
[ERROR] ^
`

katysaintin, 2023-09-04:
Finish, 

```
> mvn compile  javadoc:javadoc
[INFO] Reactor Summary:
[INFO]
[INFO] phoebus (parent) ................................... SUCCESS [  0.715 s]
[INFO] dependencies ....................................... SUCCESS [  0.015 s]
[INFO] install-jars ....................................... SUCCESS [  0.561 s]
[INFO] phoebus-target ..................................... SUCCESS [  1.082 s]
[INFO] core ............................................... SUCCESS [  0.007 s]
[INFO] core-vtype ......................................... SUCCESS [  1.465 s]
[INFO] core-formula ....................................... SUCCESS [  3.747 s]
[INFO] core-framework ..................................... SUCCESS [  8.099 s]
[INFO] core-types ......................................... SUCCESS [  1.729 s]
[INFO] core-util .......................................... SUCCESS [  3.223 s]
[INFO] core-pva ........................................... SUCCESS [ 11.986 s]
[INFO] core-pv ............................................ SUCCESS [  4.652 s]
[INFO] core-security ...................................... SUCCESS [  2.267 s]
[INFO] core-ui ............................................ SUCCESS [ 11.395 s]
[INFO] core-logbook ....................................... SUCCESS [  2.710 s]
[INFO] core-email ......................................... SUCCESS [  1.787 s]
[INFO] core-launcher ...................................... SUCCESS [  4.284 s]
[INFO] app ................................................ SUCCESS [  0.004 s]
[INFO] app-diag ........................................... SUCCESS [  2.623 s]
[INFO] app-log-configuration .............................. SUCCESS [  3.089 s]
[INFO] app-probe .......................................... SUCCESS [  2.769 s]
[INFO] app-pvtable ........................................ SUCCESS [  3.242 s]
[INFO] app-pvtree ......................................... SUCCESS [  2.950 s]
[INFO] app-update ......................................... SUCCESS [  2.427 s]
[INFO] app-email .......................................... SUCCESS [  0.005 s]
[INFO] app-email-ui ....................................... SUCCESS [  2.807 s]
[INFO] app-errlog ......................................... SUCCESS [  2.458 s]
[INFO] app-filebrowser .................................... SUCCESS [  3.001 s]
[INFO] app-logbook ........................................ SUCCESS [  0.003 s]
[INFO] app-logbook-inmemory ............................... SUCCESS [  1.836 s]
[INFO] app-logbook-ui ..................................... SUCCESS [  3.644 s]
[INFO] app-logbook-olog ................................... SUCCESS [  0.004 s]
[INFO] app-logbook-olog-client ............................ SUCCESS [  4.753 s]
[INFO] app-logbook-olog-client-es ......................... SUCCESS [  4.440 s]
[INFO] app-logbook-olog-ui ................................ SUCCESS [  4.779 s]
[INFO] app-logbook-elog ................................... SUCCESS [  2.150 s]
[INFO] app-rtplot ......................................... SUCCESS [  4.831 s]
[INFO] app-databrowser .................................... SUCCESS [  6.747 s]
[INFO] app-databrowser-timescale .......................... SUCCESS [  2.654 s]
[INFO] app-display ........................................ SUCCESS [  0.003 s]
[INFO] app-display-model .................................. SUCCESS [  6.060 s]
[INFO] app-display-representation ......................... SUCCESS [  2.639 s]
[INFO] app-3d-viewer ...................................... SUCCESS [  2.673 s]
[INFO] app-display-representation-javafx .................. SUCCESS [  4.796 s]
[INFO] app-display-fonts .................................. SUCCESS [  2.103 s]
[INFO] app-display-runtime ................................ SUCCESS [  4.156 s]
[INFO] app-display-editor ................................. SUCCESS [  4.492 s]
[INFO] app-display-convert-medm ........................... SUCCESS [  3.985 s]
[INFO] app-display-convert-edm ............................ SUCCESS [  4.317 s]
[INFO] app-display-navigation ............................. SUCCESS [  2.627 s]
[INFO] app-display-adapters ............................... SUCCESS [  2.329 s]
[INFO] app-display-thumbwheel ............................. SUCCESS [  2.558 s]
[INFO] app-display-linearmeter ............................ SUCCESS [  2.779 s]
[INFO] app-alarm .......................................... SUCCESS [  0.007 s]
[INFO] app-alarm-model .................................... SUCCESS [  2.905 s]
[INFO] app-alarm-ui ....................................... SUCCESS [  3.546 s]
[INFO] app-alarm-logging-ui ............................... SUCCESS [  2.917 s]
[INFO] app-alarm-datasouce ................................ SUCCESS [  2.420 s]
[INFO] app-scan ........................................... SUCCESS [  0.010 s]
[INFO] app-scan-model ..................................... SUCCESS [  6.155 s]
[INFO] app-scan-client .................................... SUCCESS [  2.135 s]
[INFO] app-scan-ui ........................................ SUCCESS [  3.501 s]
[INFO] app-channel ........................................ SUCCESS [  0.006 s]
[INFO] app-channel-channelfinder .......................... SUCCESS [  4.391 s]
[... 10 lines cut ...]
[INFO] app-console ........................................ SUCCESS [  2.333 s]
[INFO] app-trends ......................................... SUCCESS [  0.004 s]
[INFO] app-trends-rich-adapters ........................... SUCCESS [  2.164 s]
[... cut ...]
[INFO] service-alarm-config-logger ........................ SUCCESS [  1.853 s]
[INFO] service-save-and-restore ........................... SUCCESS [  6.672 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 04:22 min
[INFO] Finished at: 2023-09-04T23:17:58+02:00
[INFO] Final Memory: 50M/160M
[INFO] ------------------------------------------------------------------------
```

katysaintin, 2023-09-05:
I think you can close this issue now @shroffk

shroffk, 2023-09-05:
Awesome work @katysaintin

closed 2023-09-05 by shroffk