ControlSystemStudio/phoebus #2804: Save restore nodes filtering issue

jembishop, 2023-09-04:
Hi there. I was trying to use the search functionality in the Phoebus save restore to search for a snapshot of a specific name.
I noticed that when the search string was just "2023" the filtering seemed to be working correctly. However once I put in the "-" character to narrow down the date, it returns zero search results. It looks like this is a bug? It would be useful to have this fixed as it means we could easily search for a snapshot at specific time.  Thanks
[image]
[image]

georgweiss, 2023-09-04:
The observed behavior is a side effect of how the underlying search engine tokenizes text, in this case the name of a snapshot.

I recommend using the start time and end time to locate data based on date/time.

georgweiss, 2023-09-07:
@jembishop, to follow up: does my suggestion solve your problem?

jembishop, 2023-09-12:
Sorry for the delay in reply @georgweiss. Thank you for your suggestion. 

The end goal for our project is to have something where the machine physicists can:
- Enter a timestamp and see a table of some historic PVs
- After inspecting the values, restore the set of PVs to their historical values. 

We were finding this difficult to integrate with the existing way the save-restore system works for two main reasons.
Firstly the save-restore system didn't integrate with the EPICs archiver appliance, so we would have to run some job to generate these snapshots and dump them all in save restore, and another job to continually generate snapshots, and this would also lead to coarse granularity. Then the issue is having a interface where users can quickly get the timestamp they are looking for and we were relying on the search functionality to do this, but the date search you suggested will not work for the archived snapshots I think even if they are given the date as a name.

At the moment I was going the way of creating my own app to do this. I give a quick demo in this video:

https://github.com/ControlSystemStudio/phoebus/assets/40360024/7b95fafa-f5d8-453f-8989-dde17981cc3a

georgweiss, 2023-09-13:
So my understanding is that you would like to read PV values from the archiver at some given point in time and then create a snapshot from the these values. Is this correct?

jembishop, 2023-09-13:
That is the functionality we are looking for yes.

georgweiss, 2023-09-13:
Save&restore is designed to be interactive in the sense that snapshots are created in the client UI when a (sub)system has been set/tuned/calibrated to a wanted state. There is a fundamental difference between your use case and this interactive mode: snapshots generated from the UI will perform an explicit read of the PVs, while archived values may be monitor values, or even averaged values.

As the save&restore service is agnostic as to how the values in a snapshots are determined, you may of course use the service the way you describe to persist the snapshot. Once persisted it can be restored from the UI.

In the case of searching for a date I still cannot see why you could not use the start and end time to define a time span in the search. Alternatively you could select to name the snapshot in a different format, e.g. yyyyMMddHHmmss. Dashes, space char and colon are used by Elasticsearch for tokenization, so avoiding these in the name could solve your issue.

jembishop, 2023-09-22:
I will close this for now, as I think you have cleared up all my questions. Thanks for your help.

closed 2023-09-22 by jembishop