ControlSystemStudio/phoebus #2820: Save&restore roles definition

georgweiss, 2023-09-15:
For authorization purposes in save&restore service we need to define a few "roles", i.e. something of the kind:

- Unauthenticated users: browse/read data (all GET endpoints)
- Role 0 (all authenticated users): take and save snapshots. Delete should be possible if user id matches.
- Role 1: +restore snapshots
- Role 2: no restrictions. 

This would make management of configurations and composite snapshots exclusive to role 2. 

@shroffk , please share your thoughts.

georgweiss, 2023-10-03:
@shroffk, need input in order to proceed.

georgweiss, 2023-10-03:
That was the idea. At least at ESS some configurations have been created with too little attention to the contents. But frankly, if this is too limiting then could be granted to Role 0.

shroffk, 2023-10-03:
I would vote for keeping the role for creating configurations a bit more permissive to encourage users to use SAR more... I understand the issue with the SAR being cluttered with many repetitive configurations.

Just like the alarm server, we could then create a stats... this particular configuration was not used to create any snapshots or the snapshots have never been restored maybe it should be removed

georgweiss, 2023-10-03:
OK, actually agree that creating configurations should be easy, despite the price...

shroffk, 2023-10-20:
Based on feedback collected for NSLS2 and ESS users the new Roles requested are as follows

1. Creating configurations, Taking snapshots, Restoring snapshots: requires authentication but no specific role
2. . Adding tags: requires authentication but no specific role
3.  Updating Configuration, updating snapshot: requires authentication + ownership or admin role
4.  Deleting configuration, Deleting snapshots, Adding and Deleting Golden Snapshots tag: require authentication + admin role

So currently, the requirement is for an admin role which primarily limits data deletion.

georgweiss, 2023-10-25:
@shroffk, my users are not on board with this proposal. They need to separate possibility of restore - as that is the primary use case - from management of objects. The simplest solution we could agree on is:

1. Authenticated user: restore operation.
2. Authenticated + role0: Creating configurations, Taking snapshots, tagging (except golden), manage objects if ownership matches.
3. Authenticated + role1: no restrictions. A login timeout is suggested to avoid costly mistakes.

closed 2023-12-14 by georgweiss