ControlSystemStudio/phoebus #2823: Building phoebus product for windows

tanviash, 2023-09-20:
Has anyone tried building the latest Phoebus product for Windows OS? Seem to give the following error.

[ERROR] Failed to execute goal on project phoebus-target: Could not resolve dependencies for project org.phoebus:phoebus-target:jar:4.7.3-SNAPSHOT: The following artifacts could not be resolved: org.openjfx:javafx-controls:jar:windows:19, org.openjfx:javafx-graphics:jar:windows:19, org.openjfx:javafx-base:jar:windows:19, org.openjfx:javafx-web:jar:windows:19, org.openjfx:javafx-media:jar:windows:19, org.openjfx:javafx-swing:jar:windows:19, org.openjfx:javafx-fxml:jar:windows:19: Could not find artifact org.openjfx:javafx-controls:jar:windows:19 in sonatype-nexus-snapshots (https://oss.sonatype.org/content/repositories/snapshots) -> [Help 1]

georgweiss, 2023-09-20:
Successfully ran a build (on Mac), all javafx libs resolved correctly, screenshot from directory phoebus-product/target/lib. Using JDK 11.0.5.

[image]

tanviash, 2023-09-20:
@georgweiss Thanks for trying. Do you see a zip folder in phoebus-product/target/?

georgweiss, 2023-09-20:
Yes

shroffk, 2023-09-20:
I built on windows ( did not use the cross compile argument )

`mvn clean verify -DskipTests`

```
shroffk@LPS-164306 /cygdrive/c/git/cs-studio/phoebus/phoebus-product/target
$ ls -alh
total 196M
drwxrwx---+ 1 shroffk Domain Users    0 Sep 20 10:32 .
drwxrwx---+ 1 shroffk Domain Users    0 Sep 20 10:32 ..
drwxrwx---+ 1 shroffk Domain Users    0 Sep 20 10:32 antrun
drwxrwx---+ 1 shroffk Domain Users    0 Sep 20 10:32 archive-tmp
drwxrwx---+ 1 shroffk Domain Users    0 Sep 20 10:32 lib
drwxrwx---+ 1 shroffk Domain Users    0 Sep 20 10:32 maven-archiver
-rwxrwx---+ 1 shroffk Domain Users 5.5K Sep 20 10:32 product-4.7.3-SNAPSHOT.jar
-rwxrwx---+ 1 shroffk Domain Users 196M Sep 20 10:32 product-4.7.3-SNAPSHOT.zip
```

tanviash, 2023-09-20:
Right, zip folder appears when product is built for Windows on either Mac or Windows but when it is built on Linux for Windows, tar.gz gets built.

georgweiss, 2023-09-20:
@tanviash , I use same as you.

shroffk, 2023-09-20:
So there are packaging profile which are activated using the os info.... when cross compiling you will need to activate these profiles manually

So in your case tanvi you will need to use 

`mvn clean install -DskipTests=true -Djavafx.platform=win -Dmaven.javadoc.skip=true -Pwindows
`

tanviash, 2023-09-20:
Thanks Kunal and Georg. I think the issue is resolved. Closing the ticket.

closed 2023-09-20 by tanviash