ControlSystemStudio/phoebus #2825: Option to turn off IPv6

jembishop, 2023-09-21:
Hi there. we are trying to use https://github.com/ornl-epics/pvws , inside a docker container (also inside WSL2)

Our setup doesn't support IPv6, so it is failing on https://github.com/ControlSystemStudio/phoebus/blob/master/core/pva/src/main/java/org/epics/pva/client/ClientUDPHandler.java#L105

I don't think IPv6 is essential here correct? So would it be possible to have an option to have the `udp_beacon` IPv4, and to disable 
` udp_localaddr6`  (possibly if `-Djava.net.preferIPv4Stack=true` is set)? 

I can have a go at creating this pull request if you wish. Thanks

kasemir, 2023-09-21:
In https://github.com/ControlSystemStudio/phoebus/blob/master/core/pva/src/main/java/org/epics/pva/PVASettings.java
we claim that IPv6 support is enabled by listing one or more IPv6 addresses in the EPICS_PVA_ADDR_LIST (client), EPICS_PVA_NAME_SERVERS (client) or EPICS_PVAS_INTF_ADDR_LIST (server).

But as you note we do prepare some IPv6 stuff anyway, even if it's not used in the end. Makes sense to skip those steps when IPv6 isn't used.

I think at this time those settings are read via `Network.parseAddress`, https://github.com/ControlSystemStudio/phoebus/blob/52f22565b166a89412d01fafc40fe55c11f40e8f/core/pva/src/main/java/org/epics/pva/common/Network.java#L114, so you could put something in there to detect if we have any IPv6 address configured.

Or is that too obfuscated? Then just add a new `PVASettings.EPICS_PVA_ENABLE_IP6`, defaulting to true. That might be most straight forward to implement and also to understand when configuring PVA.

jembishop, 2023-09-22:
I would agree that a new setting `PVASettings.EPICS_PVA_ENABLE_IP6` would be best, otherwise its confusing to remember all the specific things you have to check to disable IPv6. Probably a good idea to make sure none of `EPICS_PVA*` variables have IPv6 addresses if EPICS_PVA_ENABLE_IP6 is set.

 I shall get to work implementing this feature. I am quite new to Java (and Phoebus) so it will take me a bit longer, and I might possibly have to ask for some guidance at some points.

kasemir, 2023-09-26:
You fixed it, thanks!

closed 2023-09-26 by kasemir