ControlSystemStudio/phoebus #2830: memory leak for pv creation

caraxlr, 2023-09-27:
Hi, previously I opened [an issue about the screen freeze when running phoebus](https://github.com/ControlSystemStudio/phoebus/issues/2770). And geogweiss suggested me to verify if there is a memory leak when creating pvs. I wonder about what's the normal process to release the pvs? Because if I create a display directly from phoebus > new display, it's normal to add the pvs but where to release them? 

Besides, the problem of the screen is that the screen may freeze or show glitches like this when running phoebus
My computer is RHEL8.8 with NVIDIA Corporation GP107GL [Quadro P620] (rev a1). Thanks a lot for your help.
[image]

georgweiss, 2023-09-28:
To clarify: my reference to PV creation and memory leaks is only within the context of scripts attached to a widget. If you do not use scripts, or not call ```PVUtil.createPV``` in scripts, then you're fine. PVs configured for widgets do **not** cause memory leaks.

Fro the attached image it looks like your issue is related to the graphical subsystem on the computer. There are some known issues with Phoebus and gtk3, so you should try launching with an additional JVM parameter, e.g.:

```java -Djdk.gtk.version=2 ....```

For the sake of comparison you could also run Phoebus and you OPIs on Windows or Mac.

caraxlr, 2023-09-28:
Thank you Georg, we don't have the memory leak problem for the scripts. I also think it may be a problem related to gpu.

caraxlr, 2023-09-29:
Hi Georg, I'm a little lost to debug the graphical subsystem. I was wondering if you had any suggestions? I wonder how the java application is related to the gpu operation?  Is it through packages like javafx? And do we solve this problem by changing jvm settings? Thanks a lot for your kind help.

kasemir, 2023-09-29:
See https://github.com/ControlSystemStudio/phoebus/wiki/JavaFX

closed 2024-02-01 by caraxlr