ControlSystemStudio/phoebus #2834: save restore mapping updates

shroffk, 2023-09-29:
Some save and restore configuration might have more than 10k pv's

This hits the default elastic limit on nested documents which would require additional configuration of the elastic cluster to increase this limit.

There might be some merit to exploring the possibility of storing the entire snapshot as a single document instead of a list of nested documents.

georgweiss, 2023-10-03:
@shroffk, can this be done painlessly using only Elastic's reindexing API? Or do we need migration code?

shroffk, 2023-10-03:
I think we will have to reindex and make some minimal code change (it is possible that all that needs to be updated is the message deserializer...but I still need to work on this a bit)