ControlSystemStudio/phoebus #2838: save restore connection to service

GDH-ISIS, 2023-10-09:
Here at Rutherford, we have a working version of the save and restore service (running with docker containers) and GUI which we have been looking at and getting feedback from users (nice work I might add). However, if we use the latest version of the GUI and save-restore, I am struggling to get the GUI to connect properly to the save and restore service - I believe possibly due to the recent change in the save-restore url. Service is running on port 8001 (docker). Rolling back to a compiled version (both GUI and service) before the new url works fine.

http://localhost:8001/save-restore/node/44bef5de-e8e6-4014-af37-b8f6c8a939a2 is working - elastic connected.

saveand restore settings.ini file are (from create create_settings_template.py (generated save-restore url I think needs to be updated))
#### Package org.phoebus.applications.saveandrestore
org.phoebus.applications.saveandrestore/sortSnapshotsTimeReversed=false
org.phoebus.applications.saveandrestore/readTimeout=5000
org.phoebus.applications.saveandrestore/search_result_page_size=30
org.phoebus.applications.saveandrestore/default_search_query=tags=golden
#### Package org.phoebus.applications.saveandrestore
org.phoebus.applications.saveandrestore/jmasar.service.url=http://localhost:8001/save-restore
org.phoebus.applications.saveandrestore/httpClient.readTimeout=1000
org.phoebus.applications.saveandrestore/httpClient.connectTimeout=1000

[image]
[image]
[image]

Any advice much appreciated.

georgweiss, 2023-10-09:
We restructured the client side code a bit to reduce the number of software modules. Side effect is that the preference defining the service URL now must be:
```org.phoebus.applications.saveandrestore.client/jmasar.service.url```
Note the addition of ```client``` in the name.

GDH-ISIS, 2023-10-09:
Sorry for not keeping up to date. The solution above worked first time. Much appreciated.

closed 2023-10-09 by GDH-ISIS