ControlSystemStudio/phoebus #2841: Rapidité d'affichage Tank Widget

XavSPM, 2023-10-12:
Hello everyone,

I am in the process of gradually migrating our interfaces from CS-Studio to Phoebus.

I am encountering an issue on one of the operator pages. This page allows us to display the current value measured by the BLMs. Here is what it looks like:

[image]

With CS-Studio, the bar graph is created using "Progress Bars" with the "Show Scale" option enabled.

However, in Phoebus, this option is no longer available for the "Progress Bars". Therefore, I have opted for "Tanks" to create the bar graphs. Nevertheless, it seems that the display speed of the "Tanks" is slower than that of the "Progress Bars". Given that the current reading can be very fluctuating, the "Progress Bars" widgets would be more suitable for our needs.

Videos:

- cs-studio :  

https://github.com/ControlSystemStudio/phoebus/assets/104097021/8831e3ea-9a60-4157-8154-94f34a186392

- phoebus : 

https://github.com/ControlSystemStudio/phoebus/assets/104097021/8cd4bb42-5d46-4342-9e28-3b81fe41434c

Would it be possible to integrate the "Show Scale" option into the "Progress Bars" as is the case in CS-Studio?

Thank you in advance.

Xavier

georgweiss, 2024-01-15:
Sorry for late feed-back.

The Progress Bar widget is essentially a wrapper around the native JavaFX ProgressBar, which does not support tick marks or scale. There are good reasons to keep it lightweight.

Recently we added the Linear Meter widget. It has more customization options compared to the Progress Bar. Please give it a try to see if it meets your needs.

XavSPM, 2024-01-25:
I had thought about using LinearMeter, but I am encountering two problems with this tool.

The first is that it does not handle a minimum value higher than the maximum value well. I am now using this feature since the modification made to ProgressBar.

The second problem is that it is less visible to the user. It would be useful to be able to fill the LinearMeter to make it more similar to ProgressBar.

The simplest solution for my use case would be to add a scale to ProgressBar and to display the PV value inside it. I have started to explore this possibility, but not being experienced in Java, and lacking time, I have not yet been able to dedicate myself to it.
[image]

georgweiss, 2024-01-25:
Note that the Progress Bar widget's view is the native JavaFX ProgressBar and therefore very lightweight. Adding UI elements makes it heavier.

As an alternative one could add Text Update elements to left or right of the Progress Bar widget to show numeric PV values. Likewise scales could be added as images, in particular if the size of the Progress Bar does not change.

@kasemir, can I please ask you to comment on this?

kasemir, 2024-01-25:
Yes, the display builder progress bar uses the JavaFX progress bar, which looks like the HTML progress bar (https://developer.mozilla.org/en-US/docs/Web/HTML/Element/progress), and those all agree that a progress bar is just a partially filled rectangle. The Qt progress bar goes a little further by adding a "24%" label (https://doc.qt.io/qt-6/qprogressbar.html), but nothing as complex as a scale with tick labels etc.

As technology evolves, we have to move to different display tools, for example from BOY on the Eclipse/SWT based CS-Studio to the display builder on the Phoebus/JavaFX based CS-Studio.  We can't keep pixel-by-pixel compatibility, see also https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility. A "progress bar" now looks like all the other progress bars. In the auto-conversion of older files, we could turn a legacy "progress bar" into a "tank" to get closer to the previous look. Maybe only do that when "Show Scale" was enabled? Or we could turn it into a "linear meter", maybe again only when "Show Scale" was enabled? I'm sorry to hear that you don't like either of those options, but feel free to contribute improvements that you find necessary.

XavSPM, 2024-01-26:
Thank you @georgweiss and @kasemir for your explanations.

I understand the importance of keeping the Progress Bar widget as lightweight as possible and staying as close to JavaFX as we can. I am fully aware that the way we use the Progress Bar is not its typical application.

@georgweiss, we had the same idea, I added text to display the PV value next to the Progress Bars. Unfortunately, there are ranges of values that prevent me from simply using an image for the scale.

[image]

However, I take this opportunity to tell you that you have done a great job with Phoebus. We find it much easier to use and more responsive for our purposes.

kasemir, 2024-01-26:
> .. great job .. easier to use and more responsive ..

Wait, what? This is the Github "Issues" section; there is no "Congratulations" section.
Thank you nonetheless, much appreciated!

closed 2024-02-28 by XavSPM