ControlSystemStudio/phoebus #2848: Change the Horizontal option to Rotation for ProgressBar

XavSPM, 2023-10-20:
Hello everyone,

We are beginning the gradual migration of our interfaces from CS-Studio to Phoebus.

I have an issue with one of our display pages. This page represents a kind of collimator with ProgressBar widgets. The representation looks like this:

[image]

In CS-Studio, to achieve this result, it was necessary to give a larger value to the "minimum" parameter than to the "maximum" parameter for the lower and left sides.

However, in Phoebus, this technique no longer works. The only solution I have found is to replace the "horizontal" parameter with a "rotation" parameter.

I am not an expert in Java. It is possible that my code may not be the most suitable. 
[https://github.com/XavSPM/phoebus/tree/ProgressBar_Rotation](url)

I have tested these modifications, and they seem to work well.

[image]

[image]

Would it be possible to consider this solution? What do you think?

Xavier

kasemir, 2023-10-20:
Instead of changing the parameter set from "Horizontal" to "Rotation", can't you try and make the min/max inversion work? The image plot axes support a "minimum" that's larger than the "maximum". The "minimum" is simply the bottom resp. left end of the scale and the "maximum" the top resp. right edge.

Since we have examples of axes supporting  the min/max inversion, and the older progress bar allowed it, it would be best for compatibility if you also support that in the new progress bar. Should actually be simpler than adding a rotation.
Just check how the "percentage" is set now, https://github.com/ControlSystemStudio/phoebus/blob/2b363aeee894fc7d1bb3d3928cebc9aad0f90669/app/display/representation-javafx/src/main/java/org/csstudio/display/builder/representation/javafx/widgets/ProgressBarRepresentation.java#L140, and add some `if (min_val > max_val)` case.

XavSPM, 2023-10-20:
I tried this approach, but I wasn't able to make the progress bar start from the other side of the widget.

I will try again next week. I must have forgotten something.

kasemir, 2023-10-20:
>  I wasn't able to make the progress bar start from the other side of the widget.

Right, I think the JavaFX progress bar always fills from the left.
To fill from the right, you need to rotate by 180.
Just like vertical res. not-horizontal already uses a 90 degree rotation.

So handle that in the representation:

If min/max are inverted, adjust the "percentage" computation AND add a 180 degree rotation to the JFX widget.
But keep that inside the representation.
The widget  model stays with "horizontal", "min", "max", and the representation turns that into the internally necessary 0, 90, 180, -90 degree rotation and percentage.

XavSPM, 2023-10-22:
I hadn't thought of this modification. It's much easier to do.

I've made the changes, and tried them out. I've also added a "propertyChanged" function so that if a user changes the min and max values, it also changes the running display.

I think I've done this correctly. But as I'm not a trained developer, there may still be some errors. 

#2850

closed 2023-10-31 by XavSPM