ControlSystemStudio/phoebus #2849: New release

jembishop, 2023-10-20:
If it is possible could a new release of Phoebus be issued? 
 I would like to have the new version so I can more easily include it as a binary dependency in https://github.com/slacmshankar/epicsarchiverap . I think since Phoebus is a maven project, and the archiver appliance is a gradle project, the option to use a remote github dependency in gradle will not work.  I need the new version as the IPv6 fix which I merged recently in https://github.com/ControlSystemStudio/phoebus/pull/2826 will also fix some issues I have in my build of the epics archiver appliance.
 Thanks.

shroffk, 2023-10-20:
We are currently working on releasing a few of the dependencies and the middle layer services
I expect next week to release epics core java, channelfinder, phoebus olog, save and restore... following which we should have a new release of Phoebus
Would you be able to wait another 10 days - 2 weeks?

jembishop, 2023-10-20:
Thanks sounds fine, thanks for getting back to me

GDH-ISIS, 2023-11-28:
Was wondering whether there was going to be a new release of Phoebus please? I am looking to use the IPV6 fix that was merged recently - mentioned above.

shroffk, 2023-11-28:
As @georgweiss mentioned... we need to do a new release of epicsCoreJava before a release of Phoebus.
A few of the PR's are also waiting for a release before they can be merged since they change the JDK dependency... we are trying to push forward as quickly as possible on this

kasemir, 2023-11-29:
>  we need to do a new release of epicsCoreJava

Is this about the already merged update to the VTable serialization https://github.com/epics-base/epicsCoreJava/pull/133, https://github.com/epics-base/epicsCoreJava/pull/134, https://github.com/epics-base/epicsCoreJava/pull/135, only need a release, or are we awaiting more changes?

georgweiss, 2023-11-29:
Maybe. We've encountered a case where labels array is of different length than the number of values columns. Since column count in IVTable is determined by length of labels/names, the serialized object does not include all data columns.

georgweiss, 2023-11-29:
Thanks for your understanding.

jembishop, 2024-01-22:
This is closed given a new release has [happened](https://github.com/ControlSystemStudio/phoebus/releases/tag/v4.7.3)

closed 2024-01-22 by jembishop