ControlSystemStudio/phoebus #2851: NPE in ChannelSearch if host is not reachable

georgweiss, 2023-10-26:
Spotted when using a gateway host that is not reachable unless I setup VPN connection:

In ```ChannelSearch``` line 203 ```addr.getAddress().getAddress()``` will return ```null```. Any other addresses will consequently be ignored. 

Suggest a ```null``` check (and log entry) for ```addr.getAddress().getAddress()``` so that other addresses are considered. Or will this have side effects?

kasemir, 2023-10-26:
Yes, checking for null and then ignoring that address is fine.

Not sure if it should be logged as a CONFIG message like the rest in there, or as a WARNING.

In your scenario, is the address in question one that's found automatically from enumerating all network interfaces, or is it explicitly added to the PVA_ADDR_LIST? Since you mention "gateway host", I assume it's specifically added, so what we have then is a matter of basically adding a bad address to the list. Well, temporarily bad, only correct while on VPN.
So is that just the CONFIG that you want/need, to be logged as such, or a WARNING because what you configured is wrong, but then only wrong at certain times?

georgweiss, 2023-10-26:
Yes, explicitly added, and one of them is "bad" due to lack of VPN connection.

georgweiss, 2023-10-26:
The gw address is "good" while in the office, but "bad" at home. Prefer null check to updating address list...

kasemir, 2023-10-26:
Then I suggest checking for null and 'continue' in that loop with a CONFIG warning that we're ignoring the address.

closed 2023-10-26 by shroffk