ControlSystemStudio/phoebus #2853: Shortcut for re-load displays, run displays and middle click for open in new tab

Mathis-Hu, 2023-10-26:
There were several shortcuts in CSS, like :
- Run a Display with F5
- Re-load a Display in editor and in runtime with Ctrl+G

We can also add the possibility of opening a display in a new tab with the middle click onto an action button with open display action (in addition to the shortcut Ctrl + Left Click that already exist for that purpose), it would give another way to do that, just like we can do it in a web browser.

kasemir, 2023-10-26:
> middle click

Middle-click is somewhat a Linux/X-only idea.
Plus it's already used to copy the PV name, since that's what EDM did.

Mathis-Hu, 2023-10-26:
> Plus it's already used to copy the PV name, since that's what EDM did.

The middle click is just when clicking on an action button, so it should not be conflictual with EDM, right ?

The goal of that shortcut is when you want to open a large number of displays, it's more fluid with middle click and there's a lot of shortcuts with Ctrl, so the user can use unwanted shortcuts.

closed 2024-01-26 by Mathis-Hu