ControlSystemStudio/phoebus #2855: Allow empty name for PVAStructure columns?

georgweiss, 2023-10-27:
We're working on a solution where Phoebus will need to write to a NTTable PV, and where the number of columns in the value field may vary, and where each column will be an array of strings. To support variable amount of columns we keep the number of columns constant and assign empty strings for columns that do not contain any data (=empty strings).

However, when trying to write such a PV ```PVAStructure``` will throw an exception (L103) if an empty name is encountered. Removing the check the write operation succeeds with the expected result, i.e. updates the PV as expected.

How about removing the check?

kasemir, 2023-10-27:
Sure, go ahead and make the PVAStructure constructor https://github.com/ControlSystemStudio/phoebus/blob/4f5663f5281009808a0cb1cd03653a3afbb79c8f/core/pva/src/main/java/org/epics/pva/data/PVAStructure.java#L97 accept 'elements' with empty name.

Should we still require a non-null name?
Maybe even change the PVAData constructor https://github.com/ControlSystemStudio/phoebus/blob/4f5663f5281009808a0cb1cd03653a3afbb79c8f/core/pva/src/main/java/org/epics/pva/data/PVAData.java#L28 to check for a null name? Or leave that as is to run into null pointer exceptions later on?

georgweiss, 2023-10-29:
Null check I think makes sense, so I can change the current check and test case.

As for PVData... there are currently no test cases to check for empty name or null. Cannot determine what a null check and IllegalArgumentException would mean in practice.

closed 2023-11-01 by georgweiss