ControlSystemStudio/phoebus #2856: Main Phoebus window wants to be on top....mostly

DuncanAtISIS, 2023-10-30:
When opening a dialog (ie from a script, using FileUtil.saveFileDialog) or popped out phoebus display (from an action), the main phoebus window (more often than not) immediately pushes the new window behind itself.

DuncanAtISIS, 2023-10-30:
Running on MS Windows 10

georgweiss, 2023-11-03:
Able to reproduce on Windows, but not Mac.
Looks like this is connected to rendering of the tooltip on the action button. Can you please confirm by testing the following:

1. Move mouse pointer away from the button quickly once button is pressed.
2. Set the tooltip of the action button to empty string

DuncanAtISIS, 2023-11-03:
Thanks for looking at this.  I tried those 2 tests and the results seem to confirm that it's related to tooltip rendering - the popped out display stayed on top.

georgweiss, 2023-11-03:
OK, thanks for testing. Not sure if this is a JavaFX bug. In any case I'll try to come with a fix.

georgweiss, 2023-11-04:
Turns out that the behavior described in this ticket is reproducible on Mac OS in a simple JavaFX application. So my suspicion about JavaFX may be incorrect.

georgweiss, 2023-11-06:
Also tested on Ubuntu (20) with Gnome: behaves like on Mac, i.e. new window is *not* pushed to background when mouse hovers over action button and tooltip is rendered.

georgweiss, 2023-12-07:
Fixed in #2868

closed 2023-12-07 by georgweiss