ControlSystemStudio/phoebus #2864: Content blurred in Navigation Tabs and Embedded Display

georgweiss, 2023-11-10:
When zooming out (e.g. to 75%) the contents of a Navigation Tabs and Embedded Display (with resize policy "no resize") get blurred. Seems this is a known issue when using a Scale transformation on ScrollPane, see https://bugs.openjdk.org/browse/JDK-8089499.

Using the workaround suggested in https://bugs.openjdk.org/browse/JDK-8089499 I am able to get rid of the blur. Drawbacks with the proposed workaround:

1. Uses reflection
2. Disables caching

Any thoughts on using the workaround?

kasemir, 2023-11-10:
So the fix is basically `setCache(false)`, except need reflection to accomplish that?
I'm fine with adding that, with comments that explain it, maybe including link to this or https://bugs.openjdk.org/browse/JDK-8089499). Eventually, we might be able to remove it again. Having to use reflection doesn't bother me at all.

As for the caching, maybe you can briefly check if it makes any discernible difference.

My biggest concern would be the use of zooming at all. Are your displays all wrongly sized so users need to zoom? Doesn't that mean that the size of fonts in menus, anything but the displays, are now too large or small?
Ideally, zooming would only be necessary in the editor, briefly zooming in to check some alignment.

georgweiss, 2023-11-10:
I am a bit bothered with reflection, but more bothered when users complain and when OPIs get blurry for no apparent reason.

Cannot see any discernible performance hit or UI lag on a moderately busy OPI.

As for the need of zooming... I find it difficult to assume anything about why and how it is used, and if use cases are valid and relevant. The feature is there, so it is used.

If there are no. objections I will add the fix to NavigationTabs and EmbeddedDisplayRepresentation. Any other place we should consider it?

kasemir, 2023-11-10:
Sure, go ahead.
The plain `Pane` as used by groups is not affected, only `ScrollPane`?

georgweiss, 2023-11-10:
Exactly, only ScrollPane. That's why EmbeddedDisplay using anything other than "no resize" does not blur the content on zooming.

georgweiss, 2023-11-13:
See #2865 and #2867

closed 2023-11-13 by georgweiss