ControlSystemStudio/phoebus #2879: User-friendly exception trace in ExceptionDetailsErrorDialog

kasemir, 2023-11-20:
The stack trace in the ExceptionDetailsErrorDialog
https://github.com/ControlSystemStudio/phoebus/blob/master/core/ui/src/main/java/org/phoebus/ui/dialog/ExceptionDetailsErrorDialog.java
 can be extremely useful to developers and administrators, but it tends to confuse end users.

A stack trace often has this form:
```
java.lang.Exception: Cannot do this or that
 at org.phoebus...
 at ...
 at ...
 at java.base/...
Caused by: ..NetException: Cannot connect to host 196.165.43.5
 at java.base/...
```

To the knowledgable this provides what-failed-where and consequences.
To end users, it says "Java error" and the "Caused By" is totally lost.

Ideas:
Somehow highlight the message of the exception and the caused-by message, so it looks more like

java.lang.Exception: **Cannot do this or that**
 at org.phoebus...
 at ...
 at ...
 at java.base/...
**Caused by: ..NetException: Cannot connect to host 196.165.43.5**

Always expand the details, since they're by default collapsed?
Or show just the messages for the exception and all caused-by when collapsed, somewhat like this:

java.lang.Exception: **Cannot do this or that**
**Caused by: ..NetException: Cannot connect to host 196.165.43.5**

and opening the "detail" will show the complete stack trace

shroffk, 2023-11-20:
I agree with the need to highlight the core issues

The ExceptionDetailsErrorDialog has a title and a message field.
Maybe we can use one of those to capture the highlighted information? Basically instead of highlighting the information and expecting users to pay attention to the highlight we simply extract that info and display it separately

[image]

```
    /** Open dialog that shows detail of error
     *
     *  <p>May be called from non-UI thread
     *
     *  @param title Title
     *  @param exception Exception
     */
    public static void openError(final String title, final Exception exception)
    {
        StringBuilder message = new StringBuilder();
        message.append(exception.getMessage()).append(System.lineSeparator());
        Throwable cause = exception.getCause();
        while(cause != null) {
            message.append(cause.getMessage()).append(System.lineSeparator());
            cause = cause.getCause();
        }
        openError(null, title, message.toString(), exception);
    }
```

kasemir, 2023-11-20:
YES, I like that.
Also, the "detail" pane is a read-only text field so you can copy the detail out and paste it into an email or get some class name into the IDE, which I find quite useful but I don't think anybody else knows that or would figure that out...

shroffk, 2023-11-20:
ah!! "copy" is certainly a very useful feature

shroffk, 2023-11-20:
making the header or the title text selectable seems a bit painful
the best I could do so far is add a custom button to the dialog which would copy the message to clipboard

[image]

not necessarily convinced this is great... we could have a similar copy stack trace button too.

kasemir, 2023-11-20:
Technically, the existing detailed stack trace "Text Area" that already allows select/copy does all I need.
.. but it's not obvious. "Copy Msg and Close" is very obvious.

Should that copy only the shorter message info, or the detailed stack trace?
Either one would be fine with me.

shroffk, 2023-11-20:
> Should that copy only the shorter message info, or the detailed stack trace?
Either one would be fine with me.

same here... should we do a coin toss (having 2 separate buttons would just make everything very busy )

shroffk, 2023-11-20:
Maybe we copy the message... since if you want you can expand the detail view and copy the whole stack trace from there

closed 2023-11-21 by kasemir