ControlSystemStudio/phoebus #2893: Non-modal dialogs

georgweiss, 2023-12-14:
I have users that find modal dialogs problematic under certain circumstances, e.g. dialog - possibly located on secondary monitor - blocks input while at the same time there is some urgency to interact with some OPI.

Making dialogs non-modal is in my view not desirable at all as one may end up in a situation where multiple dialogs have been opened, forgotten and not meaningful as the context may be unknown.

This is to request input from peer developers/users.

Previously discussed with my users is a strategy where - using appropriate UI tricks - modal dialogs can be more easily located.

shroffk, 2023-12-14:
> Previously discussed with my users is a strategy where - using appropriate UI tricks - modal dialogs can be more easily located.

+1

This is the correct solution in my opinion, not making all dialogs non-modal and turning Phoebus into EDM...

We can start with first opening the dialog in the most obvious place... after which we can look into helping locate existing dialogs.

kasemir, 2023-12-14:
For starters, modal dialogs should certainly be limited to the "edit" phase.
They're acceptable for the display editor, for editing the alarm configuration, for adding a PV to the data browser.

At runtime, the display needs to simply keep running without opening popups that require an operator reaction. That's why we don't want to support popups from the display runtime. Alarms are also never indicated via popups. You see them in the alarm table, maybe hear them in the annunciator, and you may have to acknowledge them, but that's via a click in the alarm table, not via a popup that blocks access to everything else. 

There's a gray zone for dialogs that create a log entry, file selectors to save a screenshot or open a file.

> strategy where - using appropriate UI tricks - modal dialogs can be more easily located.

That would be my leaning.
By default, JacaFX dialogs seem to appear just anywhere, including the "other" monitor.
`DialogHelper.positionDialog` was added such that dialogs can appear right next to for example the button that triggered them. That code is a little tricky, using `runLater` and then an added `forceToFront` to actually have an effect, but I think it's functional.

`DialogHelper` also supports the idea of persisting the dialog size and position, which I'm not sure is the best idea.
On one hand, it makes sense:
You open some dialog, like the one for rules. It's on top of the editor, so you move it to your preferred location.
Next time around, it opens in that location.
That sounded like a good idea, but my workflow tends to be more like this:
I work on some display on Monitor 1. Rules dialog remembers where I put it on monitor 1.
A week later, I work on a display on Monitor 2. I open the rule dialog, and can't immediately find it because it pops up on monitor 1. In a way, "always open next to the rules button" would be preferred to "remember were I last placed it"

georgweiss, 2023-12-14:
The complaints I hear are from control room staff, so edit mode is not the issue here.

Thanks @kasemir, @shroffk  for your input.

closed 2023-12-14 by georgweiss