ControlSystemStudio/phoebus #2897: Error when executing python scripts if first line contains "python"

lcaouen, 2023-12-18:
If you use a python script file and the first line is a comment with the word "python", then the script crashes.
If we remove the first line then the script works.
The problem seems to be in a standard java class (java.lang.ProcessBuilder) at line 73 in \app\display\runtime\src\main\java\org\csstudio\display\builder\runtime\script\internal\PythonGatewaySupport.java
The class seems to use the first line as an argument.
Is the functionality really used in the scripts ?
Could we add some code before to remove the first line if it's a comment and it contains the word "python" ?
If it's ok, I can provide the new source code and create a PR.

Error message :
```
2023-12-18 10:29:06 WARNING [org.phoebus.framework.jobs] Cmd PythonErrors: Traceback (most recent call last):
2023-12-18 10:29:06 WARNING [org.phoebus.framework.jobs] Cmd PythonErrors:   File "\myscript.py", line 2, in <module>
2023-12-18 10:29:06 WARNING [org.phoebus.framework.jobs] Cmd PythonErrors:     from org.csstudio.display.builder.runtime.script import PVUtil, ScriptUtil
2023-12-18 10:29:06 WARNING [org.phoebus.framework.jobs] Cmd PythonErrors: ImportError: No module named org.csstudio.display.builder.runtime.script
```

Script used
```
# Script python executed by jython
# import des packages pour gérer les PVs et widgets dans phoebus
from org.csstudio.display.builder.runtime.script import PVUtil
from org.csstudio.display.builder.runtime.script import ScriptUtil
from org.csstudio.display.builder.runtime.script import ColorFontUtil

###############################################################################
# Le script commmence ici
################################################################################

startStop = PVUtil.getString(pvs[0])
print("   startStop=" + str(startStop))
if startStop == "start":
    pvs[5].write(2)
```

kasemir, 2023-12-18:
What you found is by design. In principle, we try to support

 * Javascript
 * Jython
 * C-Python

The distinction is made via the "is..." calls in 
https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/java/org/csstudio/display/builder/model/properties/ScriptInfo.java

Javascript: Ends in "*.js" or is marked as embedded javascript code. JS tends to start up the fastest, but the javascript libraries have been changing on us. Initially, BOY used the Rhino JavaScript engine. With Java 8, we switched to the Nashorn JS engine because it was included in JDK 8. With Java 15, Nashorn gets removed from the JDK, so on 2020-06-20 we reverted back to Rhino. These changes made JS scripts incompatible over time.

Both Jython and C-Python can be in files ending in "*.py".
Jython has been supported from the start with no changes to the jython library. The Display Builder API of course changed from BOY, but overall Jython scripts have been the most stable.

C-Python is of interest because it opens up access to Numpy etc, all those tools in the C-Python ecosystem which we cannot use in Jython. We started to use the Py4J library which starts C-Python in an external process, then communicates with that in a way from the JRE such that C-Python can access the java API.

How we select between Jython and C-Python is shown in the examples like
https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/resources/examples/python/scripts_python_detail.bob
and https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/resources/examples/python/python.py,
https://github.com/ControlSystemStudio/phoebus/tree/master/app/display/model/src/main/resources/examples/scripts
vs. 
https://github.com/ControlSystemStudio/phoebus/tree/master/app/display/model/src/main/resources/examples/scripts/python

If *.py scripts contain "python" in the first line,
an external Python process will be launched to execute the script.
Otherwise we run the script in Jython.

A C-Python script can use Py4J to interact with the display via
proxy objects for the "widget", "pvs[]" and some script utility classes.

The Connect2j helper was introduced to allow writing scripts in a way that they run with both Jython and C-Python,
https://github.com/ControlSystemStudio/phoebus/tree/master/app/display/model/src/main/resources/examples/connect2j

Compared to Jython, starting an external python script and establishing the proxy objects takes longer.
Similarly, releasing the proxies on shutdown takes time.
I think it's fair to say that the C-Python support was never as practical as the Jython support, so with the display builder we kept some C-Python/Py4y examples but it's not much advertised.

Bottom line:
Of the potential scripting options (JS, C-Python, Jython), Jython should be given a preference. 
If you want to use jython, don't mention "python" in the first line.
And as a more general rule, avoid scripts if at all possible. If you do use scripts, be prepared to maintain them over changes in the display tool implementation, and be prepared to read the Java sources to learn about the API. Don't start any scripts before having read every script in the example displays.

kasemir, 2023-12-18:
If you check the initial display builder online help, https://github.com/kasemir/org.csstudio.display.builder/blob/master/org.csstudio.display.builder.editor.rcp/doc/display.builder.html, it contained a more elaborate "Python Scripts" section that explains how "python" in the first line as in
```
#!/usr/bin/env python
```

selects C-Python, how to install py4j, and how connect2j can then be used to create scripts that run both with Jython and C-Python:
```
#!/usr/bin/env python
from connect2j import scriptContext

with scriptContext('widget', 'pvs', 'ScriptUtil', 'PVUtil', dict=globals()):
    # Code that uses DisplayBuilder APIs; for example:
    value = PVUtil.getDouble(pvs[0])
```

Regrettably, the added complexity of having to install py4j and the slower startup & shutdown of the external C-Python process made this impractical. Still, the support is fundamentally still there, "python" in the first line tries to launch an external C-Python process.

lcaouen, 2023-12-18:
Thanks for the answer and all the details.
It's clearer for me now.
I close the issue.

closed 2023-12-18 by lcaouen