ControlSystemStudio/phoebus #2899: XYPlot enhancement

georgweiss, 2023-12-19:
I have a request to add the following feature in the XYPlot:

When specifying only waveform PV for Y axis, the indices of the waveform are used as X values. User requests the ability to recompute X values from indices using a scaling factor and offset.

An alternative solution could be to add a new array formula function, e.g. ```arrayIndices(<waveform PV>)```, which would return the indices as an integer array. This in turn could then be fed into the ```scale``` formula function to compute the X values.

Any thoughts?

kasemir, 2023-12-19:
The XYPlot should just display the data as received, not create/adjust the data.

New array functions could be a way to compute the necessary X axis data, but even better would be a computation on the IOC end. The IOC should produce the data that's to be displayed, because that way you can show the data with a "caget" or in any other generic EPICS client without the need to add computations into the client.

For plain IOCs, the "array calc" record might be helpful, see "aCalcoutRecord" in https://github.com/epics-modules/calc
For python-based IOCs, you can of course compute whatever you want in there already.
Note that the IOC that produces the X axis could be separate from the one which produces the Y data. Some IOC connected to control system hardware may provide the Y data, and the X axis data is then generated in a separate "GUI support" IOC.

georgweiss, 2023-12-19:
Thanks @kasemir for your input.

shroffk, 2023-12-19:
> An alternative solution could be to add a new array formula function, e.g. arrayIndices(<waveform PV>), which would return the indices as an integer array. This in turn could then be fed into the scale formula function to compute the X values

I think this would be my preference.

Like Kay mentioned... an IOC would be the best place to do this.
In CS-Studio we had various array functions which would be useful to help users figure out the kind of processing they wanted to do. Once they had tested and finalized on the behaviour the operation would be moved into an IOC.

The other benefit of implementing this in formula functions is that it will be
> easier to support in other UI's like DBWR
> easier to migrate to some future tool

georgweiss, 2023-12-19:
Thanks all. Closing this.

closed 2023-12-19 by georgweiss

kasemir, 2023-12-19:
With a "normal" IOC, PyDevice https://github.com/klemenv/PyDevice would be another way to put the computation into the IOC while at the same time being able to use python for the formula.

Kunal's right about the advantages of formula functions for the PV, but there are downsides as well:
We'll never be able to provide the full functionality of python,
our syntax will always be unique while python is now common knowledge,
and anything you add now will be work for the re-implementation in the next tool.