ControlSystemStudio/phoebus #2900: Reuse AlarmClient instances?

georgweiss, 2024-01-05:
Debugging an issue related to Alarm Table and discovered that a new AlarmClient instance is created whenever Alarm Table, Alarm Tree and Alarm Area Panel are launched, or when user switches to another alarm config.

Seems this could be optimized a bit, i.e. if an application has instantiated an AlarmClient, other applications could reuse this. Some kind of manager API would be needed to keep track of instantiated and referenced AlarmClient instances.

Feed-back appreciated.

kasemir, 2024-01-08:
Yes, in an operational scenario you likely have a few alarm client instances:
Main control room running one each for the area panel, alarm tree, alarm table, annunciator.
Same for additional control stations in the conventional facilities, cryo and target area.
So we get about a dozen alarm clients spread over 4 different hosts.

I have not noticed this as a problem when checking network traffic or profiler logs.
Kafka can handle several thousand additional clients.
We're using the scalability of kafka and keep our side of the code simple:
Each client creates an AlarmClient instance, each instance only needs to track one config and its listeners.
Sure, we could add a layer to pool instances for each config, but at this point I don't see the need.

georgweiss, 2024-01-10:
Well, the problem I am investigating _could_ be related to Kafka connection issues... But I am not quite done, so I'd like to keep this open.

georgweiss, 2024-01-25:
Closing this as my issue seems to be related to client connection to Kafka is down.

closed 2024-01-25 by georgweiss