ControlSystemStudio/phoebus #2916: Deprecation warning for Java 17

georgweiss, 2024-01-24:
So I'm looking into migration to Java 17 and get a disturbing warning during build of File Browser app:

```[WARNING] .../FileSystemPreferences.java:[79,29] [removal] AccessController in java.security has been deprecated and marked for removal```

Looks like some kind of headache...

kasemir, 2024-01-24:
In principle, we want to use the `java.util.prefs.Preferences`, but with some control over where they are stored. On Linux, the java prefs use `java.util.prefs.FileSystemPreferences` to store them in plain files, we like that. On the Mac, it could instead use `java.util.prefs.MacOSXPreferences` and I think on Windows there's also a registry-based implementation, neither of which we can control.

So we copied the Linux implementation from  `java.util.prefs.FileSystemPreferences` into `org.phoebus.framework.preferences.FileSystemPreferences`, with a slight modification to control where the pref files are located based on the `org.phoebus.framework.workbench.Locations`:

```
 // If Phoebus locations have been set,
// place the user preferences in the user directory that also stores mementos etc.
// Otherwise fall back to original FileSystemPreferences behavior
if (System.getProperty(Locations.PHOEBUS_USER, "").length() > 0)
    userRootDir = new File(System.getProperty(Locations.PHOEBUS_USER), ".userPrefs");
else
    userRootDir = new File(System.getProperty("java.util.prefs.userRoot", System.getProperty("user.home")),
        ".phoebus/.userPrefs");
```

The FileSystemPreferences which we copied contain calls to the `AccessController` which are marked as deprecated since JDK17. For what it's worth, the code still runs with JDK20.
When I check the `java.util.prefs. FileSystemPreferences` that come with JDK20, the implementation looks a little different, but still contains a deprecated `AccessController.doPrivileged` call.
So bottom line the maintainers of the JDK don't seem to have a better solution at this time.

What we could do:
 a) Nothing, since it still works with JDK20
 b) Update our `org.phoebus.framework.preferences.FileSystemPreferences` to a copy from JDK17 or 20 to be more up to date, but I don't see how that would have any effect
c) Wait for a future JDK to come with an updated `java.util.prefs.FileSystemPreferences` that no longer results in warnings. The JDK maintainers will have to provide that once the deprecated types are truly removed
d) Just give up on trying to control the java preference store? We load settings from `-settings /path/to/settings.ini` files into the Java preferences, which get persisted who knows were, doesn't matter to us?

georgweiss, 2024-01-24:
@kasemir, next LTS version is 21. Did you try that JDK?

georgweiss, 2024-01-24:
Quick test shows still compiles on 21.

kasemir, 2024-01-24:
Thanks for checking!
So we do nothing, wait for a future JDK to give us an updated `java.util.prefs.FileSystemPreferences`?

georgweiss, 2024-01-24:
Yes, maybe wait until we hit the wall and then decide for an action.

closed 2024-01-25 by georgweiss